mne-tools/mne-python #11015: Export to EEGLAB should support single and double precision

cbrnr, 2022-08-08:
Currently, exporting to EEGLAB always writes double precision floats. However, @arnodelorme mentioned in #11013, the EEGLAB default is actually single precision. Therefore, we should introduce a new parameter `precision` with two possible options `"single"` and `"double"`. I'm not sure what the default should be, probably `"double"` to be on the safe side.

This should probably be implemented in `eeglabio`. Speaking of which, I wanted to suggest to include the code in MNE-Python directly to make things easier for us. I know we discussed outsourcing this to minimize maintenance costs for us, but this is not really the case. I actually think that maintenance will be easier if we have this code in our package. WDYT?

agramfort, 2022-08-08:
> Currently, exporting to EEGLAB always writes double precision floats.
> However, @arnodelorme <https://github.com/arnodelorme> mentioned in #11013 [...]
both work for me.

> This should probably be implemented in eeglabio. Speaking of which, I
> wanted to suggest to include the code in MNE-Python directly to make things
> easier for us. I know we discussed outsourcing this [...]
I don't think it's a sustainable strategy. It also slows down releases for
eeglabio. I would suggest to have a more active set of maintainers in
eeglabio

> Message ID: ***@***.***>
>

jackz314, 2022-08-08:
I only have very limited time to maintain eeglabio due to other projects at hand, so it would be great if more people are willing to help maintain it.

cbrnr, 2022-08-09:
I guess for EEGLAB export it is sustainable because it's not as complex as e.g. BrainVision or EDF export (since the format is basically a .MAT file). It would be easier for me to make changes just in one place instead of two.

But you are right that this should probably not happen right now until the API has stabilized (because then I don't expect many breaking changes).

larsoner, 2022-10-21:
Closing for [eeglab](https://github.com/jackz314/eeglabio/issues/9)

larsoner, 2022-10-21:
https://github.com/jackz314/eeglabio/issues/9

closed 2022-10-21 by larsoner