mne-tools/mne-python #11026: plot function get error?

1358467720, 2022-08-11:
#### Describe the bug
Hi! I just load my data from edf file to plot it! It does not look like normal? Have  my data something wrong or the plot function is wrong ？
```
        
raw = mne.io.read_raw_edf(file_name)
        raw.resample(128)
        print(raw.info)
        raw.load_data()
        raw.filter(l_freq=1, h_freq=30) 
        raw.notch_filter(freqs=50)  

        raw.plot()
```
[image]

cbrnr, 2022-08-11:
Can you open and view your file with [EDFbrowser](https://www.teuniz.net/edfbrowser/) to verify that it's not the data?

1358467720, 2022-08-16:
> Can you open and view your file with [EDFbrowser](https://www.teuniz.net/edfbrowser/) to verify that it's not the data?
emm.. It does not look good with EDFbrowser. Does it mean that my data have got something wrong ?

cbrnr, 2022-08-16:
If it looks the same in EDFbrowser, then yes, something's wrong with your data and not with our import or plotting functions.

closed 2022-08-16 by cbrnr