mne-tools/mne-python #11027: Some Brain GUI widgets don't look good on Windows

hoechenberger, 2022-08-11:
I believe this screenshot illustrates my main issues:

[image]

- The toolbar is filled with white none-ness
- The "reset" rescale button is smaller than the plus and minus buttons
- The plus and minus rescale buttons look weird, the symbols are not rendered in an expected way

This is on Windows 10 with MNE 1.1.

drammock, 2022-08-11:
@alexrockhill just refactored the widgets after the 1.1 release, so there's a chance that this is already fixed on main. Can you check?

larsoner, 2022-09-15:
I think there is no way we can realistically fix this for 1.2 in the next week @hoechenberger as the proposed fix is to swap in the new Qt backend for `Brain` which is a big undertaking. Bumping the milestone...

hoechenberger, 2022-09-15:
@larsoner I actually just added it to the milestone to remind myself to check whether the problem actually still exists :) Which I'll probably do tomorrow!

larsoner, 2023-10-05:
Just to add to this, the `help` menu on macOS also covers some of the toolbar icons

[image]

hoechenberger, 2023-10-05:
The menu should now be placed inside the window, but in the menu bar (which is detached and located at the top of the screen)