mne-tools/mne-python #11036: Style of 'bool' x-ref is different than the style of other parameter types

mscheltienne, 2022-08-16:
The style for the `bool` x-ref is different than for other parameter types. See for instance the [`set_montage` method](https://mne.tools/dev/generated/mne.Epochs.html#mne.Epochs.set_montage).

So it turns out that `intersphinx` resolves `bool` to https://docs.python.org/3/library/stdtypes.html#boolean-values
and not to https://docs.python.org/3/library/functions.html#bool
which results in a different style.

I see 2 ways to fix it, both over my knowledge in sphinx-related stuff:
- fix the resolution: [`intersphinx_disabled_reftypes`](https://www.sphinx-doc.org/en/master/usage/extensions/intersphinx.html#confval-intersphinx_disabled_reftypes) seems like a good candidate
- overriding the CSS style: c.f. the note for `numpydoc_xref_param_type` on the [`numpydoc` documentation](https://numpydoc.readthedocs.io/en/latest/install.html).

larsoner, 2022-08-16:
We should be able to make an explicit link here I think

https://github.com/mne-tools/mne-python/blob/main/doc/conf.py#L178

mscheltienne, 2022-08-16:
Oh true, that's ten times easier. I'll give it a shot :)

PR #11037 (2022-08-16): Fix style of 'bool' x-ref by resolving to the class instead of the std:label
Fixes #11036
closed 2022-08-16 by drammock