mne-tools/mne-python #11044: Error when importing BIDS file

arnodelorme, 2022-08-17:
File is below. It is part of https://openneuro.org/datasets/ds002718

https://drive.google.com/file/d/18SLD5TI0R_25-0rmF0g86YODXp753uze/view?usp=sharing

```python
import mne

filename = 'ds002718/sub-002/eeg/sub-002_task-FaceRecognition_eeg.set'
raw = mne.io.read_raw_eeglab(filename)

<ipython-input-22-cf235f128b50>:5: RuntimeWarning: Unknown types found, setting as type EEG:
ekg: ['EEG063', 'EEG064']
heog: ['EEG061']
veog: ['EEG062']
  raw = mne.io.read_raw_eeglab(filename)
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
<ipython-input-22-cf235f128b50> in <module>
      3 filename = '../../data/sub-001_task-P300_run-2_eeg.set'
      4 filename = '../../ds002718/sub-002/eeg/sub-002_task-FaceRecognition_eeg.set'
----> 5 raw = mne.io.read_raw_eeglab(filename)

~/miniconda3/envs/p38env/lib/python3.8/site-packages/mne/io/eeglab/eeglab.py in read_raw_eeglab(input_fname, eog, preload, uint16_codec, verbose)
    260     .. versionadded:: 0.11.0
    261     """
--> 262     return RawEEGLAB(input_fname=input_fname, preload=preload,
    263                      eog=eog, verbose=verbose, uint16_codec=uint16_codec)
    264 

<decorator-gen-397> in __init__(self, input_fname, eog, preload, uint16_codec, verbose)

~/miniconda3/envs/p38env/lib/python3.8/site-packages/mne/io/eeglab/eeglab.py in __init__(self, input_fname, eog, preload, uint16_codec, verbose)
    366 
    367         last_samps = [eeg.pnts - 1]
--> 368         info, eeg_montage, _ = _get_info(eeg, eog=eog)
    369 
    370         # read the data

~/miniconda3/envs/p38env/lib/python3.8/site-packages/mne/io/eeglab/eeglab.py in _get_info(eeg, eog)
    183         has_pos = _eeg_has_montage_information(eeg)
    184         ch_names, ch_types, eeg_montage = \
--> 185             _get_montage_information(eeg, has_pos)
    186         update_ch_names = False
    187     else:  # if eeg.chanlocs is empty, we still need default chan names

~/miniconda3/envs/p38env/lib/python3.8/site-packages/mne/io/eeglab/eeglab.py in _get_montage_information(eeg, get_pos)
    152             if d["type"] != 'FID':
    153                 continue
--> 154             elif d['description'] == 'Nasion':
    155                 nasion = np.array([d["X"], d["Y"], d["Z"]])
    156             elif d['description'] == 'Right periauricular point':

KeyError: 'description'
```

larsoner, 2022-08-17:
Closing as dup of https://github.com/mne-tools/mne-python/issues/11039

closed 2022-08-17 by larsoner

agramfort, 2022-08-19:
@arnodelorme do you agree that these .set files are broken and that MNE is doing the right thing now?

arnodelorme, 2022-08-19:
I disagree. The "description" field is not part of the EEG.chaninfo structure so this code will fail for about half of EEGLAB dataset where this structure is defined. In the other half it is probably empty, and MNE will ignore it. The structure stores removed data channels.

agramfort, 2022-08-20:
@arnodelorme you are saying that it's ok to have points with d["type"] == 'FID' and d['description'] not present? if so how do you know which FID point is which?

arnodelorme, 2022-10-11:
d[‘labels’] should contain the channel name.

Arno

agramfort, 2022-10-11:
@arnodelorme but here we talk about fiducial points right?

arnodelorme, 2022-10-11:
If type is absent, then I would not check the description field because it is not a standard field. Then then code would work.