mne-tools/mne-python #11045: bunch of codespell errors (and false alarms)

drammock, 2022-08-17:
I suspect maybe the codespell version got updated: CIs just started (?) failing like this:

https://github.com/mne-tools/mne-python/runs/7889067340?check_suite_focus=true#step:8:31

```
Error: mne/evoked.py:275: basline ==> baseline
Error: mne/conftest.py:361: nAm ==> name
Error: mne/conftest.py:361: nAm ==> name
Error: mne/dipole.py:543: nAm ==> name
Error: mne/dipole.py:607: nAm ==> name
Error: mne/dipole.py:607: nAm ==> name
Error: mne/dipole.py:607: nAm ==> name
Error: mne/dipole.py:607: nAm ==> name
Error: mne/coreg.py:1767: enfore ==> enforce
Error: mne/simulation/source.py:107: nAm ==> name
Error: mne/preprocessing/tests/test_eeglab_infomax.py:79: comparasion ==> comparison
Error: mne/forward/tests/test_make_forward.py:427: nAm ==> name
Error: mne/decoding/ssd.py:187: ording ==> ordering
Error: mne/io/brainvision/brainvision.py:934: fom ==> from
Error: mne/io/brainvision/brainvision.py:996: fom ==> from
Error: mne/io/curry/curry.py:396: enttry ==> entry
Error: mne/io/nihon/nihon.py:395: acqusition ==> acquisition
Error: mne/utils/check.py:813: nAm ==> name
Error: mne/utils/check.py:816: nAm ==> name
Error: mne/viz/misc.py:808: nAm ==> name
Error: mne/viz/_3d.py:2866: nAm ==> name
Error: doc/conf.py:726: bu ==> by
Error: doc/_includes/institutional-partners.rst:33: bu ==> by
Error: doc/changes/0.20.inc:162: nAm ==> name
Error: doc/changes/0.14.inc:172: nAm ==> name
Error: tutorials/simulation/80_dics.py:87: nAm ==> name
Error: tutorials/inverse/80_brainstorm_phantom_elekta.py:35: nAm ==> name
Error: tutorials/inverse/80_brainstorm_phantom_elekta.py:137: nAm ==> name
Error: tutorials/inverse/80_brainstorm_phantom_elekta.py:157: nAm ==> name
Error: tutorials/stats-sensor-space/75_cluster_ftest_spatiotemporal.py:144: possbile ==> possible, possibly
Error: examples/simulation/simulate_raw_data.py:70: nAm ==> name
Error: examples/simulation/simulated_raw_data_using_subject_anatomy.py:97: nAm ==> name
Error: examples/simulation/source_simulator.py:63: nAm ==> name
Error: examples/inverse/mixed_source_space_inverse.py:178: nAm ==> name
```

most of these will go away if we special-case nAm (nanoAmperes) but some of them appear legit.

larsoner, 2022-08-17:
Yes I cut a codespell release today :)

drammock, 2022-08-18:
OK, I'll open a PR.

PR #11047 (2022-08-18): fixes for latest codespell release
closes #11045
closed 2022-08-18 by larsoner