mne-tools/mne-python #11054: Problem with documentation navigation: elements placed outside of visible area

hoechenberger, 2022-08-19:
This occurs on my system e.g. on:

- https://mne.tools/dev/generated/mne.channels.fix_mag_coil_types.html
- https://mne.tools/dev/generated/mne.channels.read_polhemus_fastscan.html

[image]

[image]

larsoner, 2022-08-19:
Can you open this on pydata-sphinx-theme? This stress not seem MNE specific as it should occur with any sufficiently long link name, and I don't think we do any relevant CSS overrides (but please check)

One solution could involve setting the overflow (I think) in CSS but it wouldn't be pretty

larsoner, 2023-04-21:
@drammock could you please transfer this to pydata-sphinx-theme? You should have permissions I think (I do not).

drammock, 2023-04-21:
cannot transfer outside an org AFAICT. at least it doesn't offer me the option to transfer to PST.  Can you transfer issues from MNE to SG or scipy?

larsoner, 2023-04-21:
Oh... no, you can't. I'll just open a new one over there

larsoner, 2023-04-21:
... actually it's fixed in `dev`, never mind! I had only looked at stable...