mne-tools/mne-python #11079: Interpolate bridge electrodes when more than 2 electrodes are bridged together

mscheltienne, 2022-08-24:
Following @alexrockhill addition of https://mne.tools/stable/auto_examples/preprocessing/eeg_bridging.html and #10587

What about the case where more than 2 electrodes are bridged together, e.g. where a trio of electrodes are bridged together?

https://github.com/mne-tools/mne-python/blob/5ee892f2ef79f451b7bdba251f4dec18e266001e/mne/preprocessing/interpolate.py#L75-L155 

`interpolate_bridged_electrodes` only works on bridged pair by pair. But if we take the example below, we can do better for the bridge ('P5', 'PO7', 'PO3', 'O1'). 

[image]

I propose 2 changes:
- handles a group of electrodes bridged together as a single bridge and computes a single virtual channel per group at the average location.
- add a warning if you use `interpolate_bridged_electrodes` where you have bridges including more than N electrodes. 

I think a good number would be N=3, a trio of electrodes is usually either a straight line on a row/column of the cap, or a triangle. If it's a row/column (e.g. PO8, PO6, PO4) the average location is actually the center electrode (PO6). If it's a triangle, the average location is more or less equidistant from the 3 electrodes. In both cases, intuitively, I think the virtual channel will improve interpolation. Above 3 electrodes, I'm not convinced. 

-------------------

To retrieve the group of electrodes, I can think of an implementation as a graph with `networkx`, something along the lines of:

```
G = nx.Graph()
for bridge in bridged_idx:
    G.add_edge(*bridge)
groups_idx = nx.connected_components(G)
```

Any other idea to avoid the dependency on `networkx`?

larsoner, 2022-08-25:
> Any other idea to avoid the dependency on networkx

You shouldn't need it for such basic functionality, you can just use

https://docs.scipy.org/doc/scipy/reference/generated/scipy.sparse.csgraph.connected_components.html

For the bridging that sounds reasonable

alexrockhill, 2022-08-25:
One thing to consider is that by providing this functionality, we're implicitly condoning doing this. I'm fine adding it but I think it's worth considering if you have a large bridged area that the data there may be very inaccurate and it might be best to just exclude that recording. At least a note I think is warranted.

alexrockhill, 2022-08-25:
Also I think you can find which ones are connected in a group in pure python pretty easy. Just iterate over each pair and if either one of the nodes in the pair is in another bridge, add that other bridge to the group. Maybe there are more efficient implementations but the number of channels is at most 256 or so so I don't think it's an issue.

mscheltienne, 2022-08-25:
@alexrockhill Yes there are ways to implement it in python, it's just that the `networkx` version was super simple.

I agree we should not be ok with large bridged areas, and a warning should be emitted if a group is larger than N (I suggest N=3 or 4). As it is now, a large bridged area is treated as any other bridge: one virtual channel is created for each bridge in the area. So overall, I think that adding a single channel is more valid, and adding a check of the size of a bridged area and a warning is an improvement towards preventing users from interpolating large bridged areas.

mscheltienne, 2022-08-25:
I'll start adding this in a PR, but I need to figure out what is happening with TP7 and T7 in #11088 first.

PR #11094 (2022-08-29): [MRG] Improve interpolation of bridged electrodes
Closes #11079 
Closes #11088

On main, `interpolate_bridged_electrodes` creates one virtual channel for each **pair** of bridged electrodes. On this PR, it creates one virtual channel for each **group** of bridged electrodes. A warning is emitted if the group includes more than 4 electrodes (not included).

On main, the virtual channel location is determined by averaging the 3 spherical coordinates of each pair of electrodes. On this PR, the location is determined by averaging the 3 cartesian coordinates of each electrode in the group. The resu [... cut]
closed 2022-08-29 by larsoner