mne-tools/mne-python #11088: Bug with transformation from cartesian to sphere to cartesian coordinates

mscheltienne, 2022-08-25:
I am not sure it's a bug, I feel like I'm missing something obvious.. Following #11079 I've been looking at the code of `interpolate_bridged_electrodes` and I ran into a weird case. I'm testing code to add virtual channels at the average location between N bridged electrodes on this recording:

[image]

I'm not going to provide that code snippet for now, I'll wait for opinions on #11079, but it creates 3 virtual channels. One at the center of the triangle (AF3, F3, F1), one very close to PO5... and a last one, between T8 and TP8 on the wrong side of the head!!

-------------

Now, consider this example, with code very similar to `interpolate_bridged_electrodes`:

```
import numpy as np
from mne import create_info
from mne.transforms import _cart_to_sph, _sph_to_cart

info = create_info(["T7", "TP7", "T8", "TP8"], sfreq=1024, ch_types="eeg")
info.set_montage("standard_1020")
montage = info.get_montage()
pos = montage.get_positions()
assert pos["coord_frame"] == "head"

ch_pos = pos["ch_pos"]

# create virtual channel midway between T7 and TP7
ch_names = ["T7", "TP7"]
sphere_positions = np.zeros((len(ch_names), 3))
for i, ch_name in enumerate(ch_names):
    sphere_positions[i, :] = _cart_to_sph(ch_pos[ch_name])
pos_virtual = _sph_to_cart(np.average(sphere_positions, axis=0))
```

`ch_pos` reads as:

```
OrderedDict([('T7',  array([-0.08598208,  0.01487164,  0.03117337])),
             ('TP7', array([-0.08675728, -0.01495098,  0.03515866])),
             ('T8',  array([0.08326137, 0.01525819, 0.03097297])),
             ('TP8', array([ 0.08362232, -0.01508567,  0.0350577 ]))])
```

Note that T7 and TP7, with odd numbers, have a **negative** x coordinate.

`pos_virtual` reads as `array([[ 8.76692427e-02, -2.68577709e-05,  3.31532296e-02]])`, with a **positive** x coordinate.

-----------

And to make it more explicit, here is the next couples of lines to get to a plot, using very similar code to `interpolate_bridged_electrodes`:

```
import numpy as np
from mne import create_info
from mne.io import RawArray
from mne.transforms import _cart_to_sph, _sph_to_cart

info = create_info(["T7", "TP7", "T8", "TP8"], sfreq=1024, ch_types="eeg")
info.set_montage("standard_1020")
montage = info.get_montage()
pos = montage.get_positions()
assert pos['coord_frame'] == "head"

ch_pos = pos["ch_pos"]

# create virtual channel midway between T7 and TP7
ch_names = ["T7", "TP7"]
sphere_positions = np.zeros((len(ch_names), 3))
for i, ch_name in enumerate(ch_names):
    sphere_positions[i, :] = _cart_to_sph(ch_pos[ch_name])
pos_virtual = _sph_to_cart(np.average(sphere_positions, axis=0))

# create virtual channel info
virtual_info = create_info(
    ch_names=["vT7-TP7"],
    sfreq=info["sfreq"],
    ch_types="eeg",
)
virtual_info["chs"][0]["loc"][:3] = pos_virtual

# create a RawArray to concatenate both info
raw = RawArray(np.random.randn(4, 1024), info)
raw_virtual = RawArray(np.random.randn(1, 1024), virtual_info)
raw.add_channels([raw_virtual], force_update_info=True)

# plot montage
raw.get_montage().plot()
```

[image]

Interestingly, changing `ch_names` to `["T8", "TP8"]` does create the correct virtual channel between T8 and TP8.

larsoner, 2022-08-25:
To me you shouldn't really need to worry about this --  you shouldn't use 2D-projected locations for meaningful data analysis (just for viz!) but rather stick with the original 3D positions.

But in any case... keep in mind in spherical coordinates, `np.average` will not be the right thing to do as it doesn't take into account wrapping. An electrode at -179deg and +179 deg are 2deg apart and all the way to the left, but you average them together and get 0 deg (on the right).

mscheltienne, 2022-08-25:
Yes, the average feels wrong here, but that's what `interpolate_bridged_electrodes` is doing at the moment.

The 2D projection I show at the end is just to illustrate; the issue is really about the 3D position. The location of the virtual channel between T7 and TP7 (both with negative `x` coordinates, left of the head) has a positive `x` coordinate (right of the head), probably because of the `average` as you mention.

mscheltienne, 2022-08-25:
If you know from the top of your head the right way to compute the average location of N electrodes in spherical coordinates, that would be great. I'm not used to working in the spherical coordinate system and it will take me a bit of time to wrap my head around it.
I'll add the fix as part of the PR for #11079.

larsoner, 2022-08-25:
My first thought was to average in cartesian, but this alone would bias the radius. So then if you convert all the old ones to spherical, compute the average *radius* only, then convert the new location to spherical, replace its radius, and convert back, it should work.

You could also use something like https://docs.scipy.org/doc/scipy/reference/generated/scipy.stats.circmean.html to just the azimuth and use the arithmetic/standard mean on the elevation and radius. That's just one idea from a quick search. Another idea is to use the average position on sphere, which would lead you to https://en.wikipedia.org/wiki/Spherical_mean, etc. That's probably the most principled way

mscheltienne, 2022-08-25:
Thanks a lot, more than enough material for me to work this one out quickly.

PR #11094 (2022-08-29): [MRG] Improve interpolation of bridged electrodes
Closes #11079 
Closes #11088

On main, `interpolate_bridged_electrodes` creates one virtual channel for each **pair** of bridged electrodes. On this PR, it creates one virtual channel for each **group** of bridged electrodes. A warning is emitted if the group includes more than 4 electrodes (not included).

On main, the virtual channel location is determined by averaging the 3 spherical coordinates of each pair of electrodes. On this PR, the location is determined by averaging the 3 cartesian coordinates of each electrode in the group. The resu [... cut]
closed 2022-08-29 by larsoner