mne-tools/mne-python #11098: Implement read method for Neurovirtual's EEG files (`.eeg`)

FelipeParreira, 2022-08-26:
#### Describe the new feature or enhancement
*Please provide a clear and concise description of what you want to add or
change.*
I need to read `.eeg` files generated by [Neurovirtual's devices](https://neurovirtual.com/br/).
Would you be able to implement a method that reads this type of file returning raw data (something similar to the `mne.io.read_raw_*` methods)?
If you are able to implement it (or at least guide me in implementing it myself), I can provide some sample files.  

#### Describe your proposed implementation
*Describe how you think the feature or improvement should be implemented (e.g., 
as a new method on an existing class? as new capability added to an existing
method?) If you're not sure, please delete this section and the next section.*

#### Describe possible alternatives
*If you've suggested an implementation above, list here any alternative
implementations you can think of, and brief comments explaining why the chosen
implementation is better.*

#### Additional comments
*Add any other context or screenshots about the feature request here.*

larsoner, 2022-08-27:
> or at least guide me in implementing it myself

That would be great!

Roughly the steps would be:

1. Add `mne/io/neurovirtual/_neurovirtual.py`
2. In this file, create a `RawNeurovirtual` that subclasses `BaseRaw`
3. Give that class a `__init__` that sets channel types and metadata, like here:

    https://github.com/mne-tools/mne-python/blob/93485e025d576470c21cc936becb3992c01a9c5e/mne/io/hitachi/hitachi.py#L231-L241

5. Give that class a `_read_segment_file` method, like here:

    https://github.com/mne-tools/mne-python/blob/93485e025d576470c21cc936becb3992c01a9c5e/mne/io/hitachi/hitachi.py#L243-L255

6. Add a `read_raw_neurovirtual(fname)` that instantiates and returns the class
7. in `mne/io/__init__.py` import `read_raw_neurovirtual`
8. Add very small test files (e.g., 1 sec of data) to [mne_testing_data](https://github.com/mne-tools/mne-testing-data#add-or-change-files-in-the-repo-for-use-with-mne)
9. Add `mne/io/neurovirtual/tests/test_neurovirtual.py` that test that data reading works

If you look at the other `mne/io/*` folders, you'll see that they all do this in various ways. A couple of simple implementations to look at would be [hitachi](https://github.com/mne-tools/mne-python/blob/93485e025d576470c21cc936becb3992c01a9c5e/mne/io/hitachi/hitachi.py), [eximia](https://github.com/mne-tools/mne-python/blob/93485e025d576470c21cc936becb3992c01a9c5e/mne/io/eximia/eximia.py), and [nicolet](https://github.com/mne-tools/mne-python/blob/93485e025d576470c21cc936becb3992c01a9c5e/mne/io/nicolet/nicolet.py).

JohnAtl, 2024-04-02:
Any updates on this?
Being able to read neurovirtual files would me very helpful.