mne-tools/mne-python #11107: Interpolating multiple bridged electrodes instead of just a pair of electrodes

adam2392, 2022-08-28:
#### Describe the new feature or enhancement
I am currently working on some EEG data in the MNE-ICALabel project and we are dealing with some subjects with bridging. I am following this [tutorial on bridging](https://mne.tools/dev/auto_examples/preprocessing/eeg_bridging.html) and am wondering if it is possible to interpolate among a set of greater than two electrodes? 

E.g. what if `C1, C2, C3, C4` are all "bridged" and I want only one virtual channel interpolated among all 4. Is this possible? 

#### Describe your proposed implementation
Looking at the `interpolate_bridged_electrodes` function, it seems this would just be the average of the signals and then the average spherical position of all 4 electrodes?

#### Additional comments
cc: @mscheltienne

larsoner, 2022-08-28:
https://github.com/mne-tools/mne-python/pull/11094 does this, right?

adam2392, 2022-08-28:
Oh did not see that!

adam2392, 2022-08-28:
Yeah it does and okay hopefully that is merged soon. I'll go take a look at that PR.

closed 2022-08-28 by adam2392