mne-tools/mne-python #11112: GSR and temperature channels do not show up in plot

hoechenberger, 2022-08-29:
MWE:
```python
# %%

from pathlib import Path
import openneuro
import mne
import mne_bids

dataset_name = 'ds003061'
bids_root = Path('/tmp') / dataset_name

# %%
openneuro.download(
    dataset=dataset_name,
    include='sub-001',
    target_dir=bids_root,
)

# %%
bp = mne_bids.BIDSPath(
    subject='001',
    task='P300',
    run='1',
    datatype='eeg',
    suffix='eeg',
    extension='.set',
    root=bids_root,
)
raw = mne_bids.read_raw_bids(bp)

with mne.viz.use_browser_backend('matplotlib'):
    raw.copy().pick(['GSR1', 'GSR2' 'Temp']).plot()
```

produces:
[image]

larsoner, 2022-08-29:
This is probably a scaling issue. There were no data to test with at the time of addition, so the `mne/defaults.py` scale factors are probably way off. In particular, the raw scale for `temperature` is almost certainly a problem here since room temperature (~20 deg C) will be off the plot given a unity `plot_raw` scale factor (the default right now IIRC)

larsoner, 2022-08-29:
Ahh I didn't even notice that. Yeah it seems like a picking problem

marsipu, 2022-08-30:
@hoechenberger Yes, when it is in matplotlib and qt-backend and no traces/y-axis-marks show I think too that it is a picking problem inside mne-python-code

larsoner, 2022-08-30:
I can try to look tomorrow

larsoner, 2022-09-08:
@hoechenberger running your example I get a bunch of junk from MNE-BIDS, followed by a real error:
```
/home/larsoner/python/mne-bids/mne_bids/read.py:60: RuntimeWarning: Unknown types found, setting as type EEG:
temp: ['Temp']
  raw = reader[ext](raw_path, **kwargs)
/home/larsoner/python/mne-bids/mne_bids/read.py:60: RuntimeWarning: Data will be preloaded. preload=False or a string preload is not supported when the data is stored in the .set file
  raw = reader[ext](raw_path, **kwargs)
/home/larsoner/python/mne-bids/mne_bids/read.py:60: RuntimeWarning: Not setting positions of 14 gsr/misc/resp channels found in montage:
['EXG1', 'EXG2', 'EXG3', 'EXG4', 'EXG5', 'EXG6', 'EXG7', 'EXG8', 'GSR1', 'GSR2', 'Erg1', 'Erg2', 'Resp', 'Plet']
Consider setting the channel types to be of EEG/sEEG/ECoG/DBS/fNIRS using inst.set_channel_types before calling inst.set_montage, or omit these channels when creating your montage.
  raw = reader[ext](raw_path, **kwargs)
/home/larsoner/python/mne-bids/mne_bids/read.py:586: RuntimeWarning: The unit for channel(s) EXG1, EXG2, EXG3, EXG4, EXG5, EXG6, EXG7, EXG8, Erg1, Erg2, Plet has changed from NA to V.
  raw.set_channel_types(channel_type_bids_mne_map_available_channels)
/home/larsoner/python/mne-bids/mne_bids/read.py:586: RuntimeWarning: The unit for channel(s) Temp has changed from V to C.
  raw.set_channel_types(channel_type_bids_mne_map_available_channels)
/home/larsoner/python/mne-bids/mne_bids/dig.py:540: RuntimeWarning: There are channels without locations (n/a) that are not marked as bad: ['EXG1', 'EXG2', 'EXG3', 'EXG4', 'EXG5', 'EXG6', 'EXG7', 'EXG8', 'GSR1', 'GSR2', 'Erg1', 'Erg2', 'Resp', 'Plet', 'Temp']
  warn(f"There are channels without locations "
/home/larsoner/python/mne-bids/mne_bids/dig.py:550: RuntimeWarning: Not setting positions of 15 gsr/misc/resp/temperature channels found in montage:
['EXG1', 'EXG2', 'EXG3', 'EXG4', 'EXG5', 'EXG6', 'EXG7', 'EXG8', 'GSR1', 'GSR2', 'Erg1', 'Erg2', 'Resp', 'Plet', 'Temp']
Consider setting the channel types to be of EEG/sEEG/ECoG/DBS/fNIRS using inst.set_channel_types before calling inst.set_montage, or omit these channels when creating your montage.
  raw.set_montage(montage, on_missing='warn')
/home/larsoner/Desktop/temp.py:29: RuntimeWarning: Channel(s) ['GSR2Temp'] could not be picked, because they are not present in the info instance.
  raw.copy().pick(['GSR1', 'GSR2' 'Temp']).plot()
/home/larsoner/python/mne-python/mne/viz/_mpl_figure.py:441: UserWarning: Attempting to set identical low and high ylims makes transformation singular; automatically expanding.
  ax_vscroll.set_ylim(len(ch_order), 0)
/home/larsoner/python/mne-python/mne/viz/_mpl_figure.py:1733: UserWarning: Attempting to set identical low and high ylims makes transformation singular; automatically expanding.
  self.mne.ax_main.set_ylim(ylim)
Traceback (most recent call last):
  File "/home/larsoner/Desktop/temp.py", line 29, in <module>
    raw.copy().pick(['GSR1', 'GSR2' 'Temp']).plot()
  File "/home/larsoner/python/mne-python/mne/io/base.py", line 1529, in plot
    return plot_raw(self, events, duration, start, n_channels, bgcolor,
  File "<decorator-gen-155>", line 12, in plot_raw
  File "/home/larsoner/python/mne-python/mne/viz/raw.py", line 355, in plot_raw
    fig = _get_browser(show=show, block=block, **params)
  File "/home/larsoner/python/mne-python/mne/viz/_figure.py", line 659, in _get_browser
    fig = backend._init_browser(**kwargs)
  File "/home/larsoner/python/mne-python/mne/viz/_mpl_figure.py", line 2278, in _init_browser
    fig._update_trace_offsets()
  File "/home/larsoner/python/mne-python/mne/viz/_mpl_figure.py", line 1738, in _update_trace_offsets
    self._update_yaxis_labels()
  File "/home/larsoner/python/mne-python/mne/viz/_mpl_figure.py", line 1642, in _update_yaxis_labels
    ticklabels = self.mne.ch_names[self.mne.picks]
IndexError: arrays used as indices must be of integer (or boolean) type
```
I see in the output and a quick sanity check a few things:

1. MNE-BIDS not setting the `Temp` channel as a temperature channel
2. Missing comma between `'GSR2'` and `'Temp'` causing them not to be picked
3. Us not handling empty picks correctly, which I'm fixing
4. The data appearing broken on read: `raw.get_data('temperature').max()` is `-5.005000233650207e-07`

I'll open a WIP PR to fix the picking bug (3), but two of the other things (1, 4) probably need to be fixed at the MNE-BIDS/OpenNeuro end I think.

PR #11150 (2022-09-13): parse temperature from ch names in .EDF files
use

```
import mne
from mne.datasets.sleep_physionet.age import fetch_data

(raw_fname, _), = fetch_data(subjects=[0], recording=[1])
raw = mne.io.read_raw_edf(raw_fname, infer_types=True)
print(raw.info)
print(raw.ch_names)
print(raw[-2, :30][0])
```

to check it works.

Closes #11112

@larsoner you will see that the values are around 37C°.
closed 2022-09-13 by agramfort