mne-tools/mne-python #11113: Scaling is off for tiny_bids

hoechenberger, 2022-08-29:


sappelhoff, 2022-08-29:
This was added to the testing data here: https://github.com/mne-tools/mne-testing-data/pull/49

As far as I remember I recorded this data in the lab at my institute and truncated it. No idea why the scaling is off 🤔

hoechenberger, 2022-08-29:
> No idea why the scaling is off 🤔

Could you try to read the data with BV Analyzer? I suspect our reader is the problem here…

sappelhoff, 2022-08-30:
@alisenz -- do you have time to:

1. download the BrainVision data in question (called `bv_dig_test`, see: https://github.com/mne-tools/mne-testing-data/tree/master/montage)
2. Go to our lab and check
    3. whether you can read that data with BrainVision Analyzer
    4. It's installed on the recording PC
    5. The dongle is in the materials room (small red USB stick, should be on top right of shelf in a porcelain bowl)
6. Report here whether you:
    7. can read and plot/view the data
    8. the units "seem right", i.e., that the voltages are in a reasonable range, unlike what Richard reports above when reading the data with MNE-Python
    
? 

No worries if you don't find the time.

alisenz, 2022-09-07:
@hoechenberger @sappelhoff the data is readable but if you expect the amplitude of 300-500 µV, this is **NOT** the case. I'm seeing min 5000 µV here:

[image]

sappelhoff, 2022-09-07:
wow, what the hell did I do back then? At least this confirms that the issue is with the data and not with our readers. Perhaps I did some weird scaling in a misguided attempt to obscure/anonymize the data, because I thought the severe cropping wasn't enough? After all this was originally only intended for a check of montages and the Brain Products CapTrak system.

Thanks for checking @alisenz

hoechenberger, 2022-09-07:
If we're off by 3 orders of magnitude it's probably nV instead of uV?

sappelhoff, 2022-09-08:
I don't think so. Knowing myself from 3 years ago I think this is most likely what happened (~80% sure):

> Perhaps I did some weird scaling in a misguided attempt to obscure/anonymize the data, because I thought the severe cropping wasn't enough?

fixing this would involve:

- re-writing the `.eeg` file with sensible data (leaving vhdr and vmrk intact), e.g., using [pybv](https://github.com/bids-standard/pybv)
- updating the accompanying fif file: [`bv_dig_raw.fif`](https://github.com/mne-tools/mne-testing-data/blob/master/montage/bv_dig_raw.fif)
- tiny_bids dataset in mne-bids needs to be re-written
- mne-tests don't need to be updated, because only the locs are being checked: https://github.com/mne-tools/mne-python/blob/9e67f7ce9100bbac452b990541e8f36a072f098a/mne/channels/tests/test_montage.py#L1021-L1028

hoechenberger, 2022-09-13:
Thank you @sappelhoff for fixing this!

closed 2022-09-13 by hoechenberger