mne-tools/mne-python #11119: `add_channels` should update `_orig_units` attribute

sappelhoff, 2022-08-30:
[`add_channels`](https://mne.tools/stable/generated/mne.io.Raw.html?highlight=add_channels#mne.io.Raw.add_channels) should add the unit the channel is in to `raw._orig_units`.

This came up in:

- https://github.com/mne-tools/mne-bids/pull/1059

 cc @hoechenberger 

```python
# %%
# Imports
import mne
import numpy as np

# Load data for testing
data_path = mne.datasets.testing.data_path()
vhdr = data_path / "montage" / "bv_dig_test.vhdr"
raw = mne.io.read_raw_brainvision(vhdr, preload=True)

# This has 67 units (as many as there are channels)
assert len(raw._orig_units) == len(raw.ch_names)

# Create a new channel and add it
new_info = mne.create_info(
    ch_names=["NewChannel"], sfreq=raw.info["sfreq"], ch_types=["gsr"]
)
new_data = np.atleast_2d(np.array([2.1e-6] * len(raw.times)))
new_channel = mne.io.RawArray(data=new_data, info=new_info, first_samp=raw.first_samp)
raw.add_channels([new_channel], force_update_info=True)

# The units of the new channel are not added to _orig_units
msg = f"failed: {len(raw._orig_units)} != {len(raw.ch_names)}"
assert len(raw._orig_units) == len(raw.ch_names), msg
```

larsoner, 2022-08-30:
I can try to look tomorrow

agramfort, 2022-08-31:
why not but it means we need to be consistent that the `_orig_units` attribute is a part the API of Raw. So I feel is not an isolated change.

sappelhoff, 2022-09-13:
@agramfort you mean making it a public attribute? I think I'd keep it private and just include this small fix

PR #11160 (2022-09-13): BUG: Fix consistency bug with orig units
Closes #11119

1. Make `raw._orig_units` always a dict (often empty), easier than dealing with `is not None` (just check for emptiness if needed)
2. Fix bug where `_orig_units` was not updated when picking/dropping channels or when adding channels

No need for `latest.inc` update because this is a private param only used AFAIK by `MNE-BIDS`. Maybe someday we should make this public somehow. But it seems like the correct fix is really perhaps to use the channel info about units and/or cals better someday.
closed 2022-09-13 by agramfort