mne-tools/mne-python #11128: Plot window not showing up

Liam-Noah, 2022-09-02:
### Description of the problem

Hello,

I'm following the tutorial here https://mne.tools/stable/auto_tutorials/intro/10_overview.html#sphx-glr-auto-tutorials-intro-10-overview-py . At one point I have to plot the raw fif file 
raw.plot(duration=5, n_channels=30)
The window appears a few milliseconds then dissapears. 
With verbose level to debug I get only this in the command prompt:

Using matplotlib as 2D backend.
Opening raw browser...
Using normal I/O
Using normal I/O

On the other hand raw.plot_psd(fmax=50) works correctly. The windows appears and is clickable. 
On pycharm both windows appear but they are images and not clickable. 

### Steps to reproduce

```Python
import mne
import matplotlib
print(matplotlib.rcParams['backend'])

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = (sample_data_folder / 'MEG' / 'sample' /
                        'sample_audvis_filt-0-40_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)

print(raw)
print(raw.info)

raw.plot_psd(fmax=50)
print('here1')
raw.plot(duration=25.0, n_channels=30, use_opengl=False, verbose='DEBUG')
print('here2')
```

### Expected results

The plot should show up in a new window.

### Actual results

The plot appears a few milliseconds then dissapears. 

### Additional information

Platform:         Windows-10-10.0.19041-SP0
Python:           3.9.0 (tags/v3.9.0:9cf6752, Oct  5 2020, 15:34:40) [MSC v.1927 64 bit (AMD64)]
Executable:       C:\Users\vladi\AppData\Local\Programs\Python\Python39\python.exe
CPU:              Intel64 Family 6 Model 94 Stepping 3, GenuineIntel: 4 cores
Memory:           15.9 GB

mne:              1.1.1
numpy:            1.23.2 {}
scipy:            1.9.0
matplotlib:       3.5.3 {backend=QtAgg}

sklearn:          1.1.2
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           1.4.3
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             2.2.0 {PyQt5=5.15.2}
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

Liam-Noah, 2022-09-02:
installing sklearn solved the issue

larsoner, 2022-09-02:
Okay, glad it's working!

closed 2022-09-02 by larsoner