mne-tools/mne-python #11131: concatenation of annotated epochs

virvw, 2022-09-03:
### Describe the new feature or enhancement

Allow for concatenation of annotated epochs, e.g. when epochs are baselined, then cropped and then concatenated (instead of baselined, concatenated and then cropped)

Usefulness: speed of computation for long epos can be reduced during stat analysis on specified time windows and for instance, excluding the baseline

### Describe your proposed implementation

Preserve the useful "(baseline period was cropped after baseline correction)" annotation implemented by @hoechenberger allowing to baseline and then crop

### Describe possible alternatives

N/A

### Additional context

Current error message is very helpful and explicit as to the nature of the issue so kuddos for the code and informative error labeling 👍 

[<EpochsFIF |  45 events (all good), 0 - 1 sec, baseline -0.2 – 0 sec (baseline period was cropped after baseline correction), ~42.5 MB, data loaded, with metadata,
  '11': 45>,
  
 C:\Users\vv221713\AppData\Local\Temp\ipykernel_1360\3541098965.py:1: RuntimeWarning: Concatenation of Annotations within Epochs is not supported yet. All annotations will be dropped.
data_epo = mne.concatenate_epochs([epos_list[ei] for ei, e in enumerate(epos_list)
Traceback (most recent call last):

agramfort, 2022-09-04:
we still do not support keep annotations at the epochs level when
concatenating. It remains to be implemented.

Now I am not sure how it relates to baseline correction. Can you share a
small code snippet to make the problem
clearer to us? 🙏

> Message ID: ***@***.***>
>

virvw, 2022-09-04:
Sure, here is a snippet,hth:

#%%============================================================
epos_list   = []

for b in tmp.metadata['Bins'].unique().sort_values():
                blah = tmp[query.format(b)].filter(foi_list[foi][0],
foi_list[foi][1],
                            l_trans_bandwidth ='auto', h_trans_bandwidth
='auto',
                            method ='iir', phase ='zero')

                if meas == '_ipow_':
                    blah.apply_hilbert(envelope = True, n_jobs = -1)
                else:
                    blah.apply_hilbert(envelope = False)

                if apply_bsln == True:
                    blah.apply_baseline(baseline = (bsln_tmin,bsln_tmax))

                blah.crop(tmin=lin_tmin, tmax=lin_tmax)

                epos_list.append(blah)
#____________________________________________________
pred_vars = [whichmeasure, 'intercept']
data_epo = mne.concatenate_epochs([epos_list[ei] for ei, e in
enumerate(epos_list)
                                               if bidx == ['1', '2', '3',
'4', '5']]

# concatenate error here

Virginie van Wassenhove <https://brainthemind.com/>
*My working hours may not be yours, respond in your own time*

agramfort, 2022-09-04:
do you really need annotations or just metadata? It's different things.
if you don't need the annotations you should remove them before doing the concatenations.

maybe blah.set_annotations(None) ?

virvw, 2022-09-04:
This does the trick
The annotation keeps track of the order of previous processing (set by
Richard) so if unnecessary, this will do
+1

Virginie van Wassenhove <https://brainthemind.com/>
*My working hours may not be yours, respond in your own time*

closed 2022-09-04 by agramfort