mne-tools/mne-python #11133: Bug with plot_topomap and coordinate transformations following #11081

mscheltienne, 2022-09-03:
### Description of the problem

Following #11081, topomaps created with digitization points provided as `Info` don't look the same.

### Steps to reproduce

```Python
from mne.datasets import testing
from mne.io import read_raw_fif

from mne.viz import plot_topomap

directory = testing.data_path() / "MEG" / "sample"
fname = directory / "sample_audvis_trunc_raw.fif"
raw = read_raw_fif(fname, preload=False)
raw = raw.crop(0, 10).pick("eeg").load_data().apply_proj()

plot_topomap(raw.get_data()[:, 101], raw.info)
```

### Expected results

On `mne-stable 1.1.1`, a nice looking topographic map:

[image]

### Actual results

On `mne-main 1.2`, a weird looking topographic map:

[image]

### Additional information

-

agramfort, 2022-09-03:
Can you share the info?

mscheltienne, 2022-09-03:
It's the MNE sample dataset "MEG" from which I picked only the EEG sensors. In the example above, I used the `testing` dataset instead of `sample`, but it's the same. e.g.

```
from mne.datasets import sample

directory = sample.data_path() / "MEG" / "sample"
fname = directory / "sample_audvis_filt-0-40_raw.fif"
```

With the filtered file above, the info representation is:

```
<Info | 15 non-empty values
 bads: 1 items (EEG 053)
 ch_names: EEG 001, EEG 002, EEG 003, EEG 004, EEG 005, EEG 006, EEG 007, ...
 chs: 60 EEG
 custom_ref_applied: False
 dev_head_t: MEG device -> head transform
 dig: 146 items (3 Cardinal, 4 HPI, 61 EEG, 78 Extra)
 file_id: 4 items (dict)
 highpass: 0.1 Hz
 hpi_meas: 1 item (list)
 hpi_results: 1 item (list)
 lowpass: 40.0 Hz
 meas_date: 2002-12-03 19:01:10 UTC
 meas_id: 4 items (dict)
 nchan: 60
 projs: Average EEG reference: on
 sfreq: 150.2 Hz
>
```

agramfort, 2022-09-04:
@mscheltienne I am lost. I see a PR in pycrostates.

what is the bug to fix?

mscheltienne, 2022-09-04:
Sorry, if that was not clear. Let's start over ;)

A while ago, I fixed a bug in `mne-python`, `sphere='eeglab'` was not working. PR: https://github.com/mne-tools/mne-python/pull/11081 But that PR had a side-effect that we did not notice at the time, namely, it messed up how the plot looks (c.f. the 2 screenshots and the code snippet with the MNE sample dataset in the first post). 

I noticed this visualization issue because the CI testing `pycrostates` against `mne-main` started failing. As far as this CI is concerned, there are 2 issues: one in `pycrostates` (an oversight that I'll fix) and one in `mne-python`, the visualization issue I described here. Thus the x-ref since `pycrostates` fix depends on this one as well. 

------------

I did not dig up enough yet to figure out why the topoplot is messed up now that https://github.com/mne-tools/mne-python/pull/11081 has been merged.

agramfort, 2022-09-04:
ok I'll let you look into it.

> Message ID: ***@***.***>
>

mscheltienne, 2022-12-15:
Closing, as this is not a bug but an unforeseen bug fix by #11081.
Previously (MNE 1.1), if `sphere=None` was provided; the default sphere size 0.095 was used. 
After (MNE 1.2), the sphere is set to `'auto'` and is fit to the electrode location. As the sample dataset does not have template but true EEG sensor location, the auto fit yields a different head-circle position, which is the correct behavior and was ignored.

closed 2022-12-15 by mscheltienne