mne-tools/mne-python #11135: UnicodeDecodeError for new version

MrNobodyCali, 2022-09-04:
### Description of the problem

When I read some .edf files by `read_raw_edf()` function, the codes from Version 0.24.1 succeed to read all the files. However, the codes from Version 1.1.1 succeed to read some files, but report an error while reading one file as:
> UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc8 in position 34915709: invalid continuation byte

### Steps to reproduce

```Python
> self.raw_data = read_raw_edf(raw_data_path, preload=False)[:][0]
where `raw_data_path` includes a .edf file.
```

### Expected results

Succeed to read all files.

### Actual results

  File "/home/xxx/anaconda3/lib/python3.7/site-packages/mne/io/edf/edf.py", line 1387, in read_raw_edf
    verbose=verbose)
  File "<decorator-gen-229>", line 12, in \_\_init\_\_
  File "/home/xxx/anaconda3/lib/python3.7/site-packages/mne/io/edf/edf.py", line 158, in \_\_init\_\_
    onset, duration, desc = _read_annotations_edf(tal_data[0])
  File "/home/xxx/anaconda3/lib/python3.7/site-packages/mne/io/edf/edf.py", line 1590, in _read_annotations_edf
    triggers = re.findall(pat, tals.decode('utf8'))
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xc8 in position 34915709: invalid continuation byte

### Additional information

Platform:         Linux-4.15.0-188-generic-x86_64-with-debian-buster-sid
Python:           3.7.11 (default, Jul 27 2021, 14:32:16)  [GCC 7.5.0]
Executable:       /home/xxx/anaconda3/bin/python
CPU:              x86_64: 256 cores
Memory:           755.8 GB

mne:              1.1.1
numpy:            1.20.3 {blas=mkl_rt, lapack=mkl_rt}
scipy:            1.7.3
matplotlib:       3.5.1 {backend=agg}

sklearn:          0.24.2
numba:            0.54.1
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           1.3.5
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             2.0.1 {PyQt5=5.9.6}
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

agramfort, 2022-09-04:
or changelog https://mne.tools/stable/whats_new.html says:

   -

   Annotations contained in EDF files are correctly read as UTF-8 according
   to the EDF specification (#10963
   <https://github.com/mne-tools/mne-python/issues/10963> by Clemens Brunner
   <https://github.com/cbrnr>)

So I would conclude that the EDF file you have does not follow the EDF file
standard but I am not an EDF expert

how were these files written?

Message ID: ***@***.***>
>

MrNobodyCali, 2022-09-04:
这是来自QQ邮箱的自动回复邮件。
 
您好呀~我是陈俊儒，我已收到您的邮件~我会尽快查阅的。如果您不认识我，请核对下您的发送地址。

MrNobodyCali, 2022-09-06:
I have exported these .edf files from original data by the `Neuro Workbench` software (from `NIHON KOHDEN`). The transformation is executed in the software and I did not use explicit codes to do this.

MrNobodyCali, 2022-09-06:
I am not sure that whether the commercial software still follows old encoding type `latin-1`, which causes the error report.

agramfort, 2022-09-06:
do you have the problem with other EDF readers?

> Message ID: ***@***.***>
>

cbrnr, 2022-09-12:
According to the [EDF specification](https://www.edfplus.info/specs/edfplus.html#annotation), TAL channels are encoded as UTF-8. Previously, we used ISO 8859-1 (aka Latin-1), which worked with files that are apparently very common in practice, but are not compliant with the standard.

I prefer sticking to the standard UTF-8 encoding, but we could think about introducing a new parameter to let users change the annotation encoding if this is really common. Alternatively, we could also `try`/`except` UTF-8 and Latin-1 encodings, which would neither require a new parameter nor any user input.

agramfort, 2022-09-12:
I would be ok with a new param to specify the encoding. Maybe
`tal_encoding` or just `encoding`?

> Message ID: ***@***.***>
>

cbrnr, 2022-09-12:
Fine with me. I think `encoding="utf8"` would be a good choice if we document it that this refers to annotation channels (other encodings cannot and should not be changed since e.g. header encoding is limited to ASCII).

agramfort, 2022-09-12:
works for me.

Can you take care of the PR? thx a lot

> Message ID: ***@***.***>
>

cbrnr, 2022-09-12:
> Can you take care of the PR?

Sure, will do!

MrNobodyCali, 2022-09-13:
@agramfort @cbrnr Thanks a lot for considering this situation. I will use the new function later.

PR #11154 (2022-09-13): Add encoding parameter to read_raw_{eb}df()
Fixes #11135. This PR adds a new parameter `encoding` to `read_raw_edf()` and `read_raw_bdf()` (as well as the underlying `RawEDF` class). Users can set a custom encoding of their annotation channel(s), which by default (and according to the EDF+ standard) is UTF8, but many files in the wild seem to ignore the standard and use Latin-1.
closed 2022-09-13 by larsoner