mne-tools/mne-python #11142: Cannot re-download sample dataset

drammock, 2022-09-07:
In attempting to replicate the issue in [this forum post](https://mne.discourse.group/t/error-failed-to-download-sample-dataset/5537) I tried the following:

```py
import mne
mne.datasets.sample.data_path(force_update=True)
```
This failed, with:
```
PermissionError: [Errno 13] Permission denied: '/data/mne_data/MNE-sample-data/subjects/sample/mri/brain-neuromag/sets/COR.fif'
```

it turns out that `COR.fif` had read-only permissions. Simply extracting the downloaded `.tar.gz` to a folder on my desktop, it ends up with RW permissions.  I have no memory of changing the permisisons on `COR.fif` manually, so my first assumption is that this is how they were shipped the last time I downloaded them from OSF.  It's easy enough for me to `rm -rf` the old folder...  but it's troublesome that `force_update` didn't work and I wonder if it's just me or if others are / will be affected if/when they try to update the sample dataset. 

Can a couple of you try to re-download the sample data in this way, and report back?  If it's a more widespread issue than just me, what is the right fix?  Maybe delete the old files first before extracting, so there isn't an attempted write on the existing read-only file?

drammock, 2022-09-07:
BTW, the dataset's `version.txt` said that the version I had was 0.11, and the version attempting to install is 0.12

larsoner, 2022-09-07:
Probably a bug with 0.11. I had 0.12 already and was able to `force_update=True` just now

larsoner, 2022-09-07:
> Maybe delete the old files first before extracting, so there isn't an attempted write on the existing read-only file?

Yes manually deleting them should work. Then you should be able to do a couple of `force_update=True`s and they should work

drammock, 2022-09-07:
> > Maybe delete the old files first before extracting, so there isn't an attempted write on the existing read-only file?
> 
> Yes manually deleting them should work.

What I meant was "should we update the downloader code to *automatically* delete the old files first?"  Not "will *manually* deleting first work."

larsoner, 2022-09-07:
I think our code is already trying to delete the old files and failing, though. In this case I think the user might need to manually delete/override. I'm not sure we can do it from Python, and even if we can, I'm not sure we should. It seems like better practice to make sure we distribute our datasets in a way that they can be deleted from Python without any force/ignore-permissions stuff

larsoner, 2022-09-07:
... actually it looks like we only manually delete for testing and misc:

https://github.com/mne-tools/mne-python/blob/b8ba02fb74f63fef08f5386ea27a0f660342c99b/mne/datasets/_fetch.py#L267-L272

I think we maybe used to for other datasets before the `pooch` transition. I guess now we rely on Pooch to do the right thing here. So maybe an upstream bug?

But really I'm not user `rmtree(..., ignore_errors=True)` would help us here, the file would still be there when `pooch` went to do its thing. And I like that `pooch` is the one that does the removal, so if you for example cancel your download, the files haven't been deleted yet. (I just checked `version.txt` while my `force_update=True` was running for example.)

drammock, 2022-09-07:
IIRC the reason for `rmtree` here is that `misc` and `testing` extract into folders that have a version number incorporated, and stripping that while extracting was not something pooch could do cleanly.  (this may have since been fixed in Pooch, not sure).

I tend to agree that we should avoid deleting first if possible, and that we should let pooch handle the deletion (when it can). So I guess we just... wait for people to complain, then tell them "manually delete the folder and try again"?  Feel free to close this issue if you agree.

larsoner, 2022-09-07:
Yes hopefully it's a transient issue and we don't make our datasets read-only. Or if we do (which maybe we should!), we figure out how to handle it reasonably with the downloaders

closed 2022-09-07 by larsoner