mne-tools/mne-python #11161: Empty plot_alignment for fsaverage and eegbci sample

nikmedoed, 2022-09-13:
### Description of the problem

I execute the first 2 blocks of code from [this article](https://mne.tools/stable/auto_tutorials/forward/35_eeg_no_mri.html), and get an empty display in jupyter notebook.
[image]

When running with CLI, the window closes instantly. 

I also got empty plotting with some other commands, but not always. What could this be related to?

### Steps to reproduce

```Python
import os.path as op
import numpy as np

import mne
from mne.datasets import eegbci
from mne.datasets import fetch_fsaverage

fs_dir = fetch_fsaverage(verbose=True)
subjects_dir = op.dirname(fs_dir)
subject = 'fsaverage'
trans = 'fsaverage'  # MNE has a built-in fsaverage transformation
src = op.join(fs_dir, 'bem', 'fsaverage-ico-5-src.fif')
bem = op.join(fs_dir, 'bem', 'fsaverage-5120-5120-5120-bem-sol.fif')

raw_fname, = eegbci.load_data(subject=1, runs=[6])
raw = mne.io.read_raw_edf(raw_fname, preload=True)

new_names = dict(
    (ch_name,
     ch_name.rstrip('.').upper().replace('Z', 'z').replace('FP', 'Fp'))
    for ch_name in raw.ch_names)
raw.rename_channels(new_names)

montage = mne.channels.make_standard_montage('standard_1005')
raw.set_montage(montage)
raw.set_eeg_reference(projection=True)  # needed for inverse modeling

mne.viz.plot_alignment(
    raw.info, src=src, eeg=['original', 'projected'], trans=trans,
    show_axes=True, mri_fiducials=True, dig='fiducials')
```

### Expected results

I want to compute forward operator for my only EEG (Brain Vision) data.
And then plot LORETA view.

### Actual results

Disappointment

### Additional information

Platform:         Windows-10-10.0.19044-SP0
Python:           3.9.12 (main, Apr  4 2022, 05:22:27) [MSC v.1916 64 bit (AMD64)]
Executable:       C:\Users\sirne\miniconda3\python.exe
CPU:              Intel64 Family 6 Model 60 Stepping 3, GenuineIntel: 8 cores
Memory:           15.9 GB

mne:              1.1.1
numpy:            1.23.1 {}
scipy:            1.9.1
matplotlib:       3.5.2 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.1.2
numba:            0.56.2
nibabel:          4.0.2
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           1.4.3
pyvista:          0.36.1 {OpenGL 4.3.0 - Build 20.19.15.5070 via Intel(R) HD Graphics 4600}
pyvistaqt:        0.9.0
ipyvtklink:       Not found
vtk:              9.1.0
qtpy:             2.0.1 {PyQt5=5.15.2}
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

nikmedoed, 2022-09-14:
I've test with pyvista
- 0.35.2
- 0.36.1
- 0.36.0
- 0.35.1
- 0.34.2

didn't help

nikmedoed, 2022-09-14:
> Can you update pyvista or otherwise use the previous minor version?

I've test with pyvista
- 0.35.2
- 0.36.1
- 0.36.0
- 0.35.1
- 0.34.2

didn't help

larsoner, 2022-09-14:
Have you tried setting any of the `set_3d_options` before making your plot?

https://mne.tools/stable/generated/mne.viz.set_3d_options.html

In particular the `antialias=False, depth_peeling=False, multi_samples=1` might work. If it does, then maybe some subset of those options is good enough.

drammock, 2022-09-14:
> When running with CLI, the window closes instantly.

If you're doing something like `python my_mne_script.py` then you should do `python -i my_mne_script.py` instead.  This will keep the python interpreter open (and thus also any plots generated by your script).

nikmedoed, 2022-09-14:
> antialias=False, depth_peeling=False, multi_samples=1
Thanks, it works.

> you should do `python -i my_mne_script.py`
Thanks, it works. I have never felt the need for this flag before.

Everything worked, thanks to everyone.

closed 2022-09-14 by nikmedoed