mne-tools/mne-python #11162: getting chpi information crashes when no hpi information are available

eort, 2022-09-14:
### Describe the new feature or enhancement

The new ( #11122 ) method `mne.chpi.get_active_chpi` calls `mne.chpi. get_chpi_info` to determine the number of active hpi coils. In case no information on hpis are available, the default behavior of `get_chpi_info` is raising an error. This is not ideal for all situations (see for example  [mne-bids #1067](https://github.com/mne-tools/mne-bids/pull/1067). Therefore it would be nice to be able to modify that behavior. 

### Describe your proposed implementation

I would say adding the `on_missing` keyword to  `mne.chpi.get_active_chpi` and passing it on to `mne.chpi. get_chpi_info` should be easy and sufficient. 

### Describe possible alternatives

alternatively, one could set the default behavior of the on_missing field to something less restrictive than raise. Not sure whether that would cause problems elsewhere.

agramfort, 2022-09-15:
I would be fine having mne.chpi.get_active_chpi just return 0 if no CHPI
are present. It seems the most logical to me.

> Message ID: ***@***.***>
>

eort, 2022-09-16:
@larsoner, I think this needs to be re-opened. I came across another problem in the implementation. Adding the `on_missing` keyword works to get rid of the error mesage, when hpi information miss, but it doesn't solve the actual problem of how the method deals with missing information. 

For example, testing it with the sample data results in a 
`TypeError: ufunc 'bitwise_and' not supported for the input types, and the inputs could not be safely coerced to any supported types according to the casting rule ''safe''`
because  chpi_info[2] will be an empty array in case no chpi info can be retrieved:
~~~.python
chpi_info = get_chpi_info(raw.info, on_missing=on_missing, verbose=verbose)
chpi_ts = raw[chpi_info[1]][0].astype(int)
chpi_active = (chpi_ts & chpi_info[2][:, np.newaxis]).astype(bool)
~~~

I suggest that: 
1. we add a test case for neuromag files with missing hpi info
2. we don't ignore the error when hpi info miss, but catch it and return zeros right away without further processing:

~~~.python
try:
    chpi_info = get_chpi_info(raw.info, on_missing='raise', verbose=verbose)
except ValueError as e:
    return np.zeros_like(raw.times)
else:
    chpi_ts = raw[chpi_info[1]][0].astype(int)
    chpi_active = (chpi_ts & chpi_info[2][:, np.newaxis]).astype(bool)
    return chpi_active.sum(axis=0)  
~~~

What do you think? Not sure zeros is the best choice of return value as it would be indistinguishable whether zeros were returned due to missing info, or due to inactive chpi. But then, I don't see a reason why this would be important.

larsoner, 2022-09-16:
> because chpi_info[2] will be an empty array in case no chpi info can be retrieved:

Rather than try/except I would add a conditional for `if len(chpi_info[2])`, it's cleaner/more precise and clearer

And yes we should have a test where we run this on Neuromag data without cHPI like `testing` (which is just a cropped/smaller version of `sample`)

eort, 2022-09-16:
Yes, agreed. Will I open a new PR, or can it somehow added here?

larsoner, 2022-09-16:
Let's do a new PR, the others have been merged already

PR #11163 (2022-09-15): add on_missing keyword to mne.chpi.get_active_hpi()

#### What does this implement/fix?
Fixes #11162 by adding the `on_missing` argument to the get_active_hpi function

#### Additional information
Not sure why, but I have problems using the `%(on_missing_chpi)s` pattern, even though I just copy&pasted within that same file. So the documentation doesn't properly render yet.