mne-tools/mne-python #11172: Deprecate mne.epochs.add_channels_epochs()

hoechenberger, 2022-09-16:
Working on #11168, I found that we have a function [`add_channels_epochs()`](https://mne.tools/stable/generated/mne.epochs.add_channels_epochs.html), which apparently only does a subset of `Epochs.add_channels()`, and is used nowhere in our code (it only appears in tests).

Since

- this function is redundant (one could use the method instead),
- it only exists for Epochs and we don't have anything similar for other data types,
- it is more limited than the method

I suggest deprecating it and marking it for removal.

Thoughts?

drammock, 2022-09-16:
Based on your summary I'd say yes, but I haven't looked at the code yet. One other thing I usually check is the blame, to see how old the code is.

hoechenberger, 2022-09-16:
> One other thing I usually check is the blame, to see how old the code is.

I did that and forgot to include it in my original posting: apparently hasn't been touched for 6–9 years :)

larsoner, 2022-09-16:
Yes please, let's do this for 1.2. Let me know if you want to do it or I should @hoechenberger

hoechenberger, 2022-09-16:
Please do go ahead @larsoner

PR #11180 (2022-09-17): MAINT: Deprecate add_channels_epochs
Closes #11172
closed 2022-09-17 by larsoner