mne-tools/mne-python #11173: Exclude BaseEpochs, BaseRaw and possibly other "base" classes from API docs?

hoechenberger, 2022-09-16:
### Proposed documentation enhancement

Currently, the documentation contains several base classes which I believe never actually become relevant to users, however they show up in the searches e.g. when searching for certain method names. I think it's confusing if when searching for `add_channels`, results include not only `Raw`, `Epochs`, and `Evoked`, but also `BaseRaw` and`BaseEpochs` with essentially the same docstrings. Could we simply not render the docstrings for those classes?

drammock, 2022-09-16:
> several base classes which I believe never actually become relevant to users

The base classes are public precisely because I (as an end user) needed to use them in my scripts. So they are relevant to users. But maybe you only meant that *the redundant documentation* is not relevant? I agree with that, and would be fine if we could suppress them somehow.

hoechenberger, 2022-09-16:
> The base classes are public precisely because I (as an end user) needed to use them in my scripts (#3794).

I don't understand, it just mentions there that they were made public for easier type checking? What does that mean? Do you **actually** need the base class, or wouldn't it make more sense to introduce "actual" types precisely for the purpose of type checking? I'm thinking of something like:

```python
# mne.typing
EpochsType = mne.BaseEpochs | mne.Epochs | mne.time_frequency.EpochsTFR
EvokedType = mne.BaseEvoked | mne.Evoked | mne.time_frequency.AverageTFR
```
?

drammock, 2022-09-16:
Something like that would have worked too (and, in retrospect, would have avoided the documentation issue you're confused by).  Creating a new convenience construct (`epochs_types`) was also proposed in that thread, and @agramfort opted for making `EpochsBase` public instead.

hoechenberger, 2022-09-16:
But this was decided 6 years ago. Today, type hints & checking are finding more and more widespread use. We should therefore revisit this decision sometime in the future.

Meanwhile I'm still working on type hints for MNE-BIDS, which will be included in `typeshed` once done.

drammock, 2022-09-16:
> But this was decided 6 years ago. Today, type hints & checking are finding more and more widespread use.

yep.

> We should therefore revisit this decision sometime in the future.

sure.

> They shouldn't appear in the docs. Whether they remain public or not I don't care so much at this point.

well, if you want to make `BaseEpochs` private again and introduce `EpochsTypes`, I invite you to convince the other devs and submit a PR.  It's certainly a more elegant solution than band-aiding the doc build to selectively not show certain public API elements (although that's probably not that hard either? I think it "just" involves deleting things like this: https://github.com/mne-tools/mne-python/blame/main/doc/reading_raw_data.rst#L41-L46).

hoechenberger, 2022-09-16:
> It's certainly a more elegant solution than band-aiding the doc build to selectively not show certain public API elements

Still my question remains, does anyone actually **use** those public classes beyond type checking? If not, I don't see removing them from the API docs as a band-aid. I would remove the docs, and add a new page that says how to perform type checks using the public but (then) undocumented classes.

drammock, 2022-09-16:
https://github.com/mne-tools/mne-python/blob/fefa7dba887089a46cf5e0f8836ff6fb07be2552/tutorials/epochs/10_epochs_overview.py#L421-L426

hoechenberger, 2022-09-16:
Ok so this bit can remain then! Nice!

larsoner, 2022-10-21:
I have now heard of at least a couple other packages using BaseRaw for example to implement their own reader (e.g., https://github.com/mne-tools/mne-python/issues/11257#issuecomment-1283616130), and in some recent PR I tried to update docs for some of these classes to say "don't instatiate this directly". So I'll close but if you think there are more actions we need to take feel free to reopen @hoechenberger

closed 2022-10-21 by larsoner

hoechenberger, 2022-10-21:
I think we should not show them or exclude them from the search results at least. They show up when you search for method names for example, I believe it's confusing -- especially if the docs say, don't instantiate directly

larsoner, 2022-10-21:
To accomplish this we can maybe prevent the methods from being documented at all for `Base*` classes. I'll take a look now...

hoechenberger, 2022-10-21:
Thanks @larsoner :)