mne-tools/mne-python #11196: `include_tmax` not considered in `mne.io.Raw.crop` to check `tmax` in bounds

gemeinl, 2022-09-22:
### Description of the problem

When cropping a `Raw` to its exact duration without including `tmax`, the code crashes.
This is clearly an edge case, but I would expect this to work.
Am I wrong on this?

### Steps to reproduce

```Python
import numpy as np
import mne

n_seconds = 1
sfreq = 100
ch_names = ['1']
data = np.ones((len(ch_names), sfreq*n_seconds))
info = mne.create_info(ch_names=ch_names, sfreq=sfreq)
raw = mne.io.RawArray(data, info)
raw.crop(tmax=n_seconds, include_tmax=False)
```

### Expected results

The code should run and the raw should not change, since it is cropped for the entire data.

### Actual results

Code crashes.

### Additional information

Platform:         Linux-5.15.63-flatcar-x86_64-with-glibc2.31
Python:           3.9.7 | packaged by conda-forge | (default, Sep  2 2021, 17:58:34)  [GCC 9.4.0]
Executable:       /opt/conda/bin/python3
CPU:              x86_64: 64 cores
Memory:           251.8 GB

mne:              1.1.dev0
numpy:            1.23.3 {}
scipy:            1.9.1
matplotlib:       3.6.0 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.1.2
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           1.5.0
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            Not found

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

agramfort, 2022-09-23:
@gemeinl with what you do you have

```python
>>> raw.times
array([0.  , 0.01, 0.02, 0.03, 0.04, 0.05, 0.06, 0.07, 0.08, 0.09, 0.1 ,
       0.11, 0.12, 0.13, 0.14, 0.15, 0.16, 0.17, 0.18, 0.19, 0.2 , 0.21,
       0.22, 0.23, 0.24, 0.25, 0.26, 0.27, 0.28, 0.29, 0.3 , 0.31, 0.32,
       0.33, 0.34, 0.35, 0.36, 0.37, 0.38, 0.39, 0.4 , 0.41, 0.42, 0.43,
       0.44, 0.45, 0.46, 0.47, 0.48, 0.49, 0.5 , 0.51, 0.52, 0.53, 0.54,
       0.55, 0.56, 0.57, 0.58, 0.59, 0.6 , 0.61, 0.62, 0.63, 0.64, 0.65,
       0.66, 0.67, 0.68, 0.69, 0.7 , 0.71, 0.72, 0.73, 0.74, 0.75, 0.76,
       0.77, 0.78, 0.79, 0.8 , 0.81, 0.82, 0.83, 0.84, 0.85, 0.86, 0.87,
       0.88, 0.89, 0.9 , 0.91, 0.92, 0.93, 0.94, 0.95, 0.96, 0.97, 0.98,
       0.99])
```

so the max time is not 1s. Even with `include_tmax=True` it complains.
Not sure how to proceed here...

gemeinl, 2022-09-23:
But the rec has a duration of 1s. I should be able to crop it to 1s, no? That it crashes with include_tmax=True is obvious, I think. It asks for one additional time when it already crashes without it.

From my point of view, if I call crop with tmax=1 and include_tmax=False, internally tmax should be reduced to tmax - 1 / raw.info["sfreq"]. Does this make sense to you?

agramfort, 2022-09-23:
can you try to open a PR about this and see if it breaks anything?

hoechenberger, 2022-09-25:
> But the rec has a duration of 1s. I should be able to crop it to 1s, no?

It appears we're off by one sample here, yes. 

btw there's a bug in the HTML repr that is not present in the text repr:

[image]

closed 2022-09-26 by larsoner