mne-tools/mne-python #11202: Incorrect raw duration displayed in HTML repr

hoechenberger, 2022-09-25:
_Originally posted by @hoechenberger in https://github.com/mne-tools/mne-python/issues/11196#issuecomment-1257090412_

btw there's a bug in the HTML repr that is not present in the text repr:

[image]

```python
import numpy as np
import mne

n_seconds = 1
sfreq = 100
ch_names = ['1']
data = np.ones((len(ch_names), sfreq*n_seconds))
info = mne.create_info(ch_names=ch_names, sfreq=sfreq)
raw = mne.io.RawArray(data, info)
```

cbrnr, 2022-09-25:
I think that's another argument for having just one repr. I don't like treating stuff separately for notebooks. At the very least, the HTML repr should take the normal repr and extend it when it makes sense, but duplicating what's already in the text repr is prone to such bugs.

hoechenberger, 2022-09-25:
Yes but I think we've agreed that your point is valid many times before, no? It's just someone has to go ahead and start implementing it already.

cbrnr, 2022-09-25:
I just wanted to make sure 😄.

agramfort, 2022-09-25:
The floor is yours Clemens

cbrnr, 2022-09-25:
Gotta find the time, but I will do it!

PR #11203 (2022-09-26): MRG: Fix displayed Raw duration in Jupyter notebook
Fixes #11202

```python
# %%
import numpy as np
import mne

n_seconds = 1
sfreq = 100
ch_names = ['1']
data = np.ones((len(ch_names), sfreq*n_seconds))
info = mne.create_info(ch_names=ch_names, sfreq=sfreq)
raw = mne.io.RawArray(data, info)
raw
```

`main`:
[image]

PR:
[image]

I'm don't feel like adding a regression test for this…
closed 2022-09-26 by larsoner