mne-tools/mne-python #11214: Add color parameter to mne.viz.plot_sensors function

timonmerk, 2022-09-29:
### Describe the new feature or enhancement

A `color` parameter being of shape `(n_sensors, color)` would be a great addition for visualization in the `mne.viz.plot_sensors`  function. 

E.g. currently in the ECoG Plotting Tutorial https://mne.tools/stable/auto_tutorials/clinical/30_ecog.html#sphx-glr-auto-tutorials-clinical-30-ecog-py only one color for all sensor locations are used by default. I looked up in the used function `_plot_sensors()` in  `mne.viz._3d.py` that the function `_plot_glyphs()` is called which uses a a single tuple color `(1.0,1.0,1.0)` for all sensor locations. 

I am not sure how easy that would be to adapt, but certainly a great addition!   

### Describe your proposed implementation

Additional function parameter in the function `mne.viz.plot_sensors` 

### Describe possible alternatives

Otherwise the `renderer` in `mne.viz._3d.py` would need to be called for every individual sensor with a single color:
 
```
renderer.sphere(center=loc[loc_idx, :], color=color, scale=scale,
                               opacity=opacity,
                               backface_culling=backface_culling)

```

agramfort, 2022-09-30:
sounds good to me

drammock, 2022-10-21:
All in attendance at today's dev meeting said this seems useful. @timonmerk are you up for doing the PR?

timonmerk, 2022-10-22:
@drammock Sounds great! I'll give it a try and reference this issue with the PR