mne-tools/mne-python #11221: [ENH]: Add support for reading Curry (Version 9) EEG files.

dengemann, 2022-10-03:
### Describe the new feature or enhancement

Currently, curry 9 files cannot be read, because its specific extensions are not recognized and supported.
To my understanding, file contents are the same. 

### Describe your proposed implementation

Indeed, a simple hack like the following solves the problem:

```
import mne
from mne.io import curry
curry.curry.FILE_EXTENSIONS['Curry 8'].update({
    'info': '.cdt.dpo',
    'labels': '.cdt.dpo'
})
```

Info and labels now have a .dpo extension.
I have a small Curry 9 file that contains no human data but only technical noise, which I could use to contribute such a patch.
Looking at the current unit tests, I feel slightly discouraged as very specific tests are performed against a pre-existing BDF file. 

### Describe possible alternatives

Adding my test file would not pass all exisiting tests and I would either need to create a Curry 9 version of the test files (unclear how to do that) or write separate tests for my test file.

How could we proceed to move ahead here? Any input would be appreciated.

agramfort, 2022-10-03:
any chance you can load the .bdf file in curry 9 and save it back to disk?

> Message ID: ***@***.***>
>

dengemann, 2022-10-03:
Let me try to explore that option. It would solve several issues actually.