mne-tools/mne-python #11225: BUG: Bug with eeglab montage

larsoner, 2022-10-05:
See

https://github.com/mne-tools/mne-python/blob/ac7cab31cd028df88bc925fe6ef68fea4403756b/mne/io/eeglab/tests/test_eeglab.py#L485-L492

It's unclear if the testing file is wrong or our reading of it, but the montage has dig points around 9m away

mmagnuski, 2022-10-26:
Hi @larsoner,
this is what I usually experience reading EEGLAB files into mne. I think this is a unit conversion problem: for example EEGLAB may use cm, but mne reads these values as m (but this is speculation, I did not confirm this).
I can take a look at some EEGLAB files I have in Matlab and see if there is any unit-specific info there.

mmagnuski, 2022-10-26:
The units I get might be stem from using EEGLAB option to get channel locations from BESA (there is such option in Edit -> Channel positions) and BESA seems to use mm for their template positions.

mmagnuski, 2022-10-26:
Yes, for my file, I can see in `EEG.chaninfo.filename` I can see that `eeglab\plugins\dipfit2.3\standard_BESA\standard-10-5-cap385.elp` was used. The file itself does not contain any information about the units, and I can't find this info in the EEG structure too.

mmagnuski, 2022-10-26:
The sample data seem to have even weirder units (sphere radius is 1):
[image]
so neither meters, cm nor inches...

larsoner, 2022-10-26:
Locations on a unit sphere perhaps?

mmagnuski, 2022-10-26:
@larsoner Yes, it seems to be the best guess. So the sample dataset does not give much hint about default channel position units.
But [this EEGLAB page](https://eeglab.org/tutorials/ConceptsGuide/coordinateSystem.html) and my own files seem to suggest mm are frequently used.

larsoner, 2022-10-26:
One easy solution here is to add a `montage_units='mm'` param to `read_raw_eeglab`. Then the user is kind of on the hook for figuring this out :)

And if you have a (hopefully short) rescaling function that you have found works really well in practice, we could add `montage_units=<float>` which means "do your best to rescale to <float> head size in meters". WDYT?

larsoner, 2022-10-26:
Want to give this `montage_units` a shot when you feel like hacking an "easy fix"? :smiley:

mmagnuski, 2022-10-26:
Sure, I can hack together a PR. What do we use for unit conversions in mne (a dict stored somewhere or some package)?

mmagnuski, 2022-10-26:
I can use a `to_meters = {'mm': .001, 'cm': 0.01}` etc., but I don't want to duplicate if we have any defaults for that.

larsoner, 2022-10-26:
We have some functions for it somewhere, let me look. `mne/io/utils.py:_check_orig_units` should get you at least the prefix part, I thought somewhere else we already had this other mapping for prefix->scale factor...

larsoner, 2022-10-26:
... that plus `mne/defaults.py:DEFAULTS['prefixes'][unit[0]]` should (almost) get you the scaling factor, you'll need something a bit more careful like

https://github.com/mne-tools/mne-python/blob/cee4e7d11116bd4e0e219fe611c0121cafd2394d/mne/io/base.py#L2112-L2130

Where you special-case `m` because it could both be a prefix or a SI unit.

This does seem like something that could use a nice `_unit_to_scale_si` function, where if you pass `m` you get `1., 'm'` and if you pass `cm` you get `0.01, 'm'` for example (i.e., multiply by this factor to get to this SI unit, given the supplied `unit`). Feel free to add this convenience function if you want. I think there are many other places we could potentially use it.

mmagnuski, 2022-10-26:
Do you think that testing if the max norm of xyz positions is within reasonable limits (let's say 5 - 15 cm?) would be good enough to warn users that the default unit ('mm') is not correct in their case. This would have to assume that head center is at 0, 0, 0.

PR #11283 (2022-10-31): Fix eeglab ch pos units
Closes #11225.

Still needs:
- [x] tests, I'll add them later
- [x] docs for the montage_unit argument
- [x] likely previous tests adaptation / style fixes
- [x] whats new
closed 2022-10-31 by larsoner