mne-tools/mne-python #11233: Browser figure ignores `matplotlib.rcParams["figure.figsize"]`

cbrnr, 2022-10-10:
I have another problem with the way MNE plotting works (or maybe it's just me doing things the wrong way). It seems like we ignore the `figure.figsize` [RC parameter](https://matplotlib.org/stable/tutorials/introductory/customizing.html), so I can't e.g. customize the created figure size with this mechanism. Which BTW is pretty important, because it is used in [Quarto](https://quarto.org/docs/computations/execution-options.html#figure-options) as a cell option for example.

It seems like the only way to set the figure size (in this particular case) is via the `MNE_BROWSE_RAW_SIZE` environment variable. This can lead to surprising results, because it leads to non-reproducible results when this setting depends e.g. on the presence of an entry in the (hidden) MNE config file.

Would it be possible to somehow use `matplotlib.rcParams["figure.figsize"]` to control our browser figure size? I'm not sure if this problem also applies to other kinds of figures as well.

agramfort, 2022-10-10:
this would apply to all figures right?

larsoner, 2022-10-10:
-1 on this, it's an unnecessary backward compat break. For years we've used this config var, and we also save it when closing the figure so that the next time you plot you get a plot of similar size (which is pretty common in UI design AFAIK, and we also do in `coreg`, etc.)

I think a better, clearer, and backward compatible way of explicitly controlling size would be to add a `size=None` param that uses the config var we've persisted when `None` and uses `size` when it's supplied. This is the way other config vars like `MNE_BROWSER_OVERVIEW_MODE` in `raw.plot` (and elsewhere config vars are supported) already work for example.

> when this setting depends e.g. on the presence of an entry in the (hidden) MNE config file.

To me the important point is not about a file being hidden, but rather at the user end that some configuration values have persistence beyond the Python session. A user shouldn't need to know or care how we persist them (e.g., in a windows registry or in some `.mne` directory). We should do a better job of advertising / describing them, and that this one and many others can be persisted and differ across systems (for good reasons), and if you want a specific behavior on all systems, you should set the corresponding `kwarg`s.

cbrnr, 2022-10-10:
> and we also save it when closing the figure so that the next time you plot you get a plot of similar size (which is pretty common in UI design AFAIK, and we also do in coreg, etc.)

Oh, and this might be the biggest problem – it gets saved even though the user never explicitly said so.

larsoner, 2022-10-10:
Yes `raw.plot(..., size=None)` indeed not there currently AFAIK

drammock, 2022-10-10:
> > and we also save it when closing the figure so that the next time you plot you get a plot of similar size (which is pretty common in UI design AFAIK, and we also do in coreg, etc.)
> 
> Oh, and this might be the biggest problem – it gets saved even though the user never explicitly said so.

Most apps on most computers do this though. When I close and re-open Firefox, it retains state in various ways, including its window size and position.  You're making it sound like what we're doing is crazy. It's not crazy, it's just not suited to what you want to do (embed a plot in a notebook).

Consider the use case that this GUI was designed for: manually browsing through raw data files looking for artifacts, for several subjects in a row.  The value of having the window size remain stable in those circumstances is significant.

cbrnr, 2022-10-10:
I certainly did not say (or intend to say) that what we are doing is crazy (in fact, what you are saying makes my use case sound like it is something crazy). Storing the window size makes sense for an interactive application, but when using it to generate a figure (e.g. in a notebook or in our documentation), a more transparent way to set the window size would be helpful. So adding a `size=None` parameter is a good idea.

agramfort, 2022-10-12:
let me pitch in here. @cbrnr let me share also my personal feeling here. A number of times I felt personally "attacked" with your requests, suggesting that decisions were made without thinking or something. Working collectively a in big project like mne requires discussions and compromise. We have very regular in person interactions with @drammock @larsoner @britta-wstnr and others. This helps a lot to step into the shoes of others.

As a big contributor to downstream projects that rely on mne, @cbrnr you are often frustrated that mne makes your life harder and I hear that. Now we need to find the right balance between upstream and downstream decisions and this takes time that I guess we should spend more together. If you want to hear my inner me, I would say that I would love to see you contributing more upstream eg contributing your GUI skills to make the mne-qt-browser even better etc.

let's find a moment to talk all together in a VC in the coming days. When is a good time for you @cbrnr ? Late afternoon is best for @larsoner and @drammock to join

cbrnr, 2022-10-12:
It's understandable to feel personally attacked when someone criticizes something that you put a lot of work in. It is sad that many(?) people here seem to interpret my contributions like that (or at least this is how it sounds to me).

Sure we can talk, only currently I'm pretty busy with the start of the semester, but in general late afternoon should be OK (only not Friday). Maybe we should also continue discussing this on Discord and not in this issue. You can also let me know how you think we could improve our communication, or what specifically I can do.

drammock, 2022-10-12:
> I am sure that there are good reasons why things are like they are, but very often I immediately get negative replies (-1, unnecessary, ...) to my suggestions (see also the discussion on duplicate plots in notebooks).

Looking at that other thread, the first reply (by me) was

> to me this feels like something that should be fixed upstream [...]

And @larsoner's first reply was

> [...] if this produces two figure displays, it's not totally clear to me that this is desired behavior or not. We should probably at least ask the MPL folks about it [...]

Are those "negative replies"?  Both acknowledge the existence of the problem; they just don't agree with your idea about *how* it should be addressed/fixed.  Both @larsoner and I gave multiple reasons (i.e. it was not arbitrary bikeshedding).

On the current thread, @larsoner did reply with "-1" and "unnecessary" but also [in the same comment](https://github.com/mne-tools/mne-python/issues/11233#issuecomment-1273419560) offered an alternative solution. Your [next](https://github.com/mne-tools/mne-python/issues/11233#issuecomment-1273433623) [two](https://github.com/mne-tools/mne-python/issues/11233#issuecomment-1273435745) responses used 112 words to push back on various other points raised by @larsoner and @agramfort, 22 words to acknowledge @larsoner's solution, and 10 words to ask a clarifying question.

A similar thing happened in the "two figures in notebooks" issue: I [put some work into figuring out the root problem](https://github.com/mne-tools/mne-python/issues/11165#issuecomment-1249707688) and asked you to take it from there; you [continued pushing your own solution](https://github.com/mne-tools/mne-python/issues/11165#issuecomment-1264296514). That pattern is, IMO, why your contributions may be interpreted as personal attacks --- you seem to spend more time refuting objections to your solution, rather than considering / trying out the alternatives that are proposed by others.

agramfort, 2022-10-13:
@drammock @cbrnr please let's talk. There is no hard feeling here.

cbrnr, 2022-10-17:
I will step back a little to think about this. Like I repeatedly said, my intention was never to attack someone, but to engage in discussion. Any yes, apparently I put more weight in pushing my own solution, but this does not mean I was attacking anyone.