mne-tools/mne-python #11240: Fixed orientations with mixed cortical and discrete source spaces?

jshanna100, 2022-10-12:
### Describe the new feature or enhancement

Not sure whether to classify this as a bug or feature request, but at the moment, building an MNE inverse operator for a mixed source space consisting of fixed orientation cortical and fixed orientation discrete source spaces is not allowed. I understand the reasoning for disallowing fixed orientation volume source spaces, but why would a discrete source space be a problem here?

### Describe your proposed implementation

The gatekeeping gets done here.

https://github.com/mne-tools/mne-python/blob/bb607d5e767ce2da068833503c5d025aba4b3ea8/mne/minimum_norm/inverse.py#L1504-L1510

### Describe possible alternatives

Possibly check explicitly against volume sources instead, rather than disallowing any mixed sources?

larsoner, 2022-10-12:
I agree in principle this should be okay. I think the check is just not specific enough -- it needs to guard against a volume being there. Surf + discrete should be okay. Are you up for making a PR to fix it @jshanna100 ?

larsoner, 2022-10-12:
I would try making the change and seeing which `pytest mne/minimum_norm` tests break. You might need to adjust some, and/or possibly also add some minimal one that would fail on `main` but pass on your PR.

jshanna100, 2022-10-12:
Local tests are no longer working for me. "make test" produces:

> python setup.py build_ext -i
running build_ext
rm -f .coverage
py.test -m 'not ultraslowtest' mne
ERROR: usage: py.test [options] [file_or_dir] [file_or_dir] [...]
py.test: error: unrecognized arguments: --cov-report=

closed 2022-10-14 by larsoner