mne-tools/mne-python #11242: BUG: ica.reject_ has no persistence

larsoner, 2022-10-12:
The `ica.reject_` param was added in https://github.com/mne-tools/mne-python/pull/5819 but it has no persistence. It also only shows up when fitting raw, despite the fact that fitting epochs you can have access to `epochs.reject` (which you should probably use for `ica.reject_` in that case). So I propose:

1. Actually save `ica.reject_`
2. Set `ica.reject_ = epochs.reject_` when fitting epochs

@agramfort does this make sense to you?

agramfort, 2022-10-13:
When you reject with ICA you don’t work to remove blinks for example. You want to remove really massive transient artifacts so it should not be the same reject as epochs. It’s not for the same purpose

hoechenberger, 2022-10-13:
I think for Epochs we should set it to None -- like you said, reject is only applied for Raw, while for epochs we have an entirely orthogonal reject mechanism in the epochs class

hoechenberger, 2022-10-13:
Unpopular proposal:

Deprecate this parameter and ask users to make use of annotate_amplitude instead

larsoner, 2022-10-13:
> Deprecate this parameter and ask users to make use of annotate_amplitude instead

I think it's a bit too disruptive to people's existing workflows

> When you reject with ICA you don’t work to remove blinks for example. You want to remove really massive transient artifacts so it should not be the same reject as epochs. It’s not for the same purpose

Coming back to this, I'm less convinced after reading our own `reject` docstring note, which says to use `epochs.drop_bad`:

> These parameters only have an effect if inst is Raw data. For Epochs, perform PTP rejection via drop_bad().

Thinking about it more, if @agramfort you're talking about *cleaned Epochs* like you'd use for Evoked analyses, I totally agree -- the `epochs.reject` there is too stringent for ICA.

But those are not the Epochs people should supply to `ica.fit` -- they should supply *uncleaned*, with rejection parameters just as relaxed as they would use when passing `reject=reject` and fitting raw.

Hence I think if people have `epochs.reject`, we should set `ica.reject_ = epochs.reject` in `ica.fit`. It's making explicit that when ICA was fit, those were the rejection criteria applied to the data.

hoechenberger, 2022-10-13:
> It's making explicit that when ICA was fit, those were the rejection criteria applied to the data.

Good argument, I'm sold.

PR #11244 (2022-10-13): BUG: Maintain ica.reject_ on I/O round-trip
Closes #11242
closed 2022-10-13 by larsoner