mne-tools/mne-python #11243: ENH: mne.set_bipolar_reference should not ignore bad channels

moritz-gerster, 2022-10-13:
### Description of the problem

When setting the bipolar reference between one good and one bad channel, the final bipolar channel is marked as good.

Bad channels are marked as bad because they contain artefacts. The bipolar reference between a bad channel and a good channel can never yield a good channel as a result. Therefore, the resulting bipolar channels should be marked as bad by mne.

[Jupyter Notebook: Steps to reproduce](https://github.com/mne-tools/mne-python/files/9771538/bipolar_ref.pdf)

### Steps to reproduce

```Python
A jupyter notebook to reproduce the problem is uploaded as a pdf.
```

### Expected results

Bipolar referencing works but the bipolar channel is automatically marked as bad.

### Actual results

Bipolar referencing works but the bipolar channel is marked as a good channel even though it is clearly a bad channel.

### Additional information

Platform:         macOS-12.6-arm64-arm-64bit
Python:           3.10.6 | packaged by conda-forge | (main, Aug 22 2022, 20:41:22) [Clang 13.0.1 ]
Executable:       /Users/moritzgerster/anaconda3/envs/bids_neu/bin/python
CPU:              arm: 10 cores
Memory:           16.0 GB

mne:              1.2.dev0
numpy:            1.23.2 {unknown linalg bindings (threadpoolctl module not found: No module named 'threadpoolctl')}
scipy:            1.9.0
matplotlib:       3.5.3 {backend=module://matplotlib_inline.backend_inline}

sklearn:          Not found
numba:            Not found
nibabel:          4.0.1
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           1.4.3
pyvista:          0.36.1 {OpenGL 4.1 Metal - 76.3 via Apple M1 Pro}
pyvistaqt:        0.9.0
ipyvtklink:       Not found
vtk:              9.1.0
qtpy:             2.2.0 {None=None}
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         0.10
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

agramfort, 2022-10-13:
agreed set_bipolar_reference should update the info['bads'] and maybe even
maybe warn that a bipolar channel is created from bad channels?

> Message ID: ***@***.***>
>

hoechenberger, 2022-10-13:
I think we should raise if a channel is bad, not just warn 🤔

larsoner, 2022-10-13:
Agreed. And to give an option we could have `on_bad='raise'` as a default, that way people can override with `'warn'` or `'ignore'` if they really know what they are doing

moritz-gerster, 2022-10-13:
If `on_bad='raise'`, which error should be raised?

larsoner, 2022-10-13:
Use the `_on_missing` handler, it will take care of these conditionals and classes for you

PR #11245 (2022-10-16): BUG: mne.set_bipolar_reference function to account for bad channels

#### Reference issue
Fixes #11243.

#### What does this implement/fix?
When setting the bipolar reference between one good and one bad channel, the final bipolar channel is currently marked as good.

Bad channels are marked as bad because they contain artifacts. The bipolar reference between a bad channel and a good channel can never yield a good channel as a result. Therefore, the resulting bipolar channels should be marked as bad by mne.

#### Scenarios
Let us assume we have anode = ['Fc5.'] and cathode=['Fc3.'] and we want [... cut]
closed 2022-10-16 by agramfort