mne-tools/mne-python #11254: I got a Complex number matrix when use mne.time_frequency.tfr.cwt

MeetXinZhang, 2022-10-16:
### Description of the problem

I got a Complex number matrix when use mne.time_frequency.tfr.cwt, here are my codes:

```
print('in: \n')
print(sample)
Ws = mne.time_frequency.morlet(sfreq=1024, freqs=np.array([10, 20]), n_cycles=2)
tfr = mne.time_frequency.tfr.cwt(X=sample, Ws=Ws, decim=1)
print('out: \n')
print(tfr)
```

in: 
```
[[-0.0171185  -0.01712345 -0.01712194 ... -0.0171574  -0.01715735
  -0.01715174]
 [-0.01471906 -0.0147273  -0.01471956 ... -0.0146988  -0.01469505
  -0.01468607]
 [-0.00083631 -0.0008417  -0.0008396  ... -0.00083318 -0.0008345
  -0.00083111]
 ...
```

out: 
```
[[[-0.01920775+0.06646499j -0.02238754+0.0662701j
   -0.02554496+0.06588112j ... -0.02560316-0.06597577j
   -0.02244102-0.06636661j -0.01925637-0.06656306j]
  [-0.01470064+0.04695463j -0.01916419+0.04640517j
   -0.02350233+0.04531965j ... -0.02355393-0.04539447j
   -0.01920823-0.04648302j -0.01473679-0.04703457j]]

 [[-0.01651212+0.05711284j -0.01924464+0.05694498j
   -0.02195792+0.05661033j ... -0.02191849-0.0564956j
   -0.01921113-0.05682995j -0.01648451-0.05699789j]
  [-0.01263726+0.04034889j -0.0164729 +0.03987624j
   -0.02020074+0.03894294j ... -0.02017442-0.03886898j
   -0.01645344-0.03980188j -0.01262466-0.04027496j]]

 [[-0.00094912+0.00328381j -0.00110662+0.0032743j
   -0.00126305+0.00325516j ... -0.00124199-0.0032003j
   -0.00108853-0.00321915j -0.000934  -0.00322858j]
  [-0.00071738+0.00231746j -0.00093773+0.0022914j
   -0.00115204+0.00223885j ... -0.0011384 -0.00219662j
   -0.0009286 -0.00224905j -0.0007127 -0.00227563j]]

 ...
```

What i want is an array consisted of numpy.float, like input(sample).
So anyone can tell me how should I do?

### Steps to reproduce

```Python
print('in: \n')
print(sample)
Ws = mne.time_frequency.morlet(sfreq=1024, freqs=np.array([10, 20]), n_cycles=2)
tfr = mne.time_frequency.tfr.cwt(X=sample, Ws=Ws, decim=1)
print('out: \n')
print(tfr)
```

### Expected results

numpy.array that consisted of Real number

### Actual results

numpy.array that consisted of Complex number

### Additional information

Platform:         Windows-10-10.0.19045-SP0
Python:           3.9.13 (main, Aug 25 2022, 23:51:50) [MSC v.1916 64 bit (AMD64)]
Executable:       D:\Anaconda\envs\torch\python.exe
CPU:              Intel64 Family 6 Model 158 Stepping 9, GenuineIntel: 8 cores
Memory:           Unavailable (requires "psutil" package)
mne:              1.1.1
numpy:            1.21.5 {blas=mkl_rt, lapack=mkl_rt}
scipy:            1.7.3
matplotlib:       3.6.0 {backend=TkAgg}

sklearn:          1.1.1
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
cupy:             Not found
pandas:           Not found
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

agramfort, 2022-10-16:
Morlet wavelet is complex valued see https://en.wikipedia.org/wiki/Morlet_wavelet

it's not a bug

agramfort, 2022-10-16:
@MeetXinZhang if you have usage questions you can use the mne forum https://mne.discourse.group

closed 2022-10-16 by agramfort