mne-tools/mne-python #11264: mne.io.read_raw_fif  Error

AdFiFi, 2022-10-20:
### Description of the problem

  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 1529, in plot
    return plot_raw(self, events, duration, start, n_channels, bgcolor,
  File "<decorator-gen-274>", line 12, in plot_raw
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\viz\raw.py", line 237, in plot_raw
    duration = min(raw.times[-1], float(duration))
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 1548, in times
    out = _arange_div(self.n_times, float(self.info['sfreq']))
TypeError: expected dtype object, got 'numpy.dtype[float64]'
python-BaseException
Backend QtAgg is interactive backend. Turning interactive mode on.

### Steps to reproduce

```Python
import mne
from mne.datasets import sample
import matplotlib.pyplot as plt
data_path = sample.data_path()
fname = data_path / 'MEG' / 'sample' / 'sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
```

### Expected results

data

### Actual results

 File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 1529, in plot
    return plot_raw(self, events, duration, start, n_channels, bgcolor,
  File "<decorator-gen-274>", line 12, in plot_raw
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\viz\raw.py", line 237, in plot_raw
    duration = min(raw.times[-1], float(duration))
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 1548, in times
    out = _arange_div(self.n_times, float(self.info['sfreq']))
TypeError: expected dtype object, got 'numpy.dtype[float64]'
python-BaseException
Backend QtAgg is interactive backend. Turning interactive mode on.

### Additional information

Platform:         Windows-10-10.0.19041-SP0
Python:           3.8.3 (default, Jul  2 2020, 17:30:36) [MSC v.1916 64 bit (AMD64)]
Executable:       C:\ProgramData\Anaconda3\python.exe
CPU:              Intel64 Family 6 Model 165 Stepping 2, GenuineIntel: 12 cores
Memory:           15.8 GB
mne:              1.2.1
numpy:            1.21.1 {OpenBLAS 0.3.17 with 12 threads}
scipy:            1.5.0
matplotlib:       3.6.1Backend QtAgg is interactive backend. Turning interactive mode on.
 {backend=QtAgg}
sklearn:          0.23.1
numba:            0.50.1
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           1.0.5
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             1.9.0 {PyQt5=5.9.7}
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0
mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

drammock, 2022-10-20:
Your traceback does not match your posted code snippet (the snippet does not have a call to `raw.plot()` in it, but your traceback suggests that that's when the error arose).  Please edit your issue to (1) include a *textual, prose description* of the problem under "description of the problem" and edit the code snippet so that if you copy-paste it into a fresh python interpreter, you get the posted traceback.

AdFiFi, 2022-10-20:
The error occurred when load the data by executing "raw = mne.io.read_raw_fif(fname))"

AdFiFi, 2022-10-20:
The error occurred when loading the data by executing "raw = mne.io.read_raw_fif(fname))"

AdFiFi, 2022-10-20:
I'm confused.
see

  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\fiff\raw.py", line 482, in read_raw_fif
    return Raw(fname=fname, allow_maxshield=allow_maxshield,
  File "<decorator-gen-384>", line 12, in __init__
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\fiff\raw.py", line 109, in __init__
    self.set_annotations(raws[0].annotations, emit_warning=False)
  File "<decorator-gen-348>", line 12, in set_annotations
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 633, in set_annotations
    new_annotations.crop(0, self.times[-1] + delta,
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 1548, in times
    out = _arange_div(self.n_times, float(self.info['sfreq']))
TypeError: expected dtype object, got 'numpy.dtype[float64]'
python-BaseException
Backend QtAgg is interactive backend. Turning interactive mode on.

drammock, 2022-10-20:
OK, that is a different traceback than the one you posted above.

AdFiFi, 2022-10-20:
OK

>>> import mne
>>> from mne.datasets import sample
>>> import matplotlib.pyplot as plt
>>> data_path = sample.data_path()
>>> fname = data_path / 'MEG' / 'sample' / 'sample_audvis_raw.fif'
>>> raw = mne.io.read_raw_fif(fname)
Opening raw data file C:\Users\13987\mne_data\MNE-sample-data\MEG\sample\sample_audvis_raw.fif...
    Read a total of 3 projection items:
        PCA-v1 (1 x 102)  idle
        PCA-v2 (1 x 102)  idle
        PCA-v3 (1 x 102)  idle
    Range : 25800 ... 192599 =     42.956 ...   320.670 secs
Ready.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\fiff\raw.py", line 482, in read_raw_fif
    return Raw(fname=fname, allow_maxshield=allow_maxshield,
  File "<decorator-gen-268>", line 12, in __init__
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\fiff\raw.py", line 109, in __init__
    self.set_annotations(raws[0].annotations, emit_warning=False)
  File "<decorator-gen-232>", line 12, in set_annotations
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 633, in set_annotations
    new_annotations.crop(0, self.times[-1] + delta,
  File "C:\ProgramData\Anaconda3\lib\site-packages\mne\io\base.py", line 1548, in times
    out = _arange_div(self.n_times, float(self.info['sfreq']))
TypeError: expected dtype object, got 'numpy.dtype[float64]'
>>>

AdFiFi, 2022-10-20:
[image]

drammock, 2022-10-20:
I suspect this is a problem with Numba; your version of numba is a few years old (v0.50, current is 0.56). can you try this?

```py
import mne
mne.set_config('MNE_USE_NUMBA', 'false')
data_path = mne.datasets.sample.data_path()
fname = data_path / 'MEG' / 'sample' / 'sample_audvis_raw.fif'
raw = mne.io.read_raw_fif(fname)
```

If that works, then try upgrading Numba and removing the `set_config` line.

AdFiFi, 2022-10-20:
I tried your suggestion but the output is still the same.

AdFiFi, 2022-10-20:
Well, thank you. And this user forum is going to be helpful.

AdFiFi, 2022-10-20:
Updating Numba works!

AdFiFi, 2022-10-20:
Upgrading Numba works!

closed 2022-10-20 by AdFiFi