mne-tools/mne-python #11269: how to use matplotlib.pyplot to plot raw?

TTyb, 2022-10-21:
### Proposed documentation enhancement

I get the following results using `raw.plot()`
[image]

and then I use `matplotlib.pyplot` to draw the raw`channel=EEG C3-A2`

[image]

I check that the `EEG F3-A2` curves of these two images are completely inconsistent, So how can I use `matplotlib.pyplot` to draw the same renderings as `raw.plot()` ，thank you 

my code:
 
```
import mne
import matplotlib.pyplot as plt

raw = mne.io.read_raw_edf('../20220826/edfs/N91EDFS/210113.edf', preload=True)
raw.plot(start=60, duration=30, scalings=dict(eeg=1e-4, resp=1e3, eog=1e-4, emg=1e-7, misc=1e-1))

raw.drop_channels(['EEG ROC-A2', 'EEG LOC-A2', 'EEG F3-A2', 'EEG O1-A2', 'EMG Chin', 'ECG', 'Snoring Snore', 'Airflow', 'Resp Chest', 'Resp Abdomen', 'SaO2 SpO2', 'Pulse', 'Pos Body_Pos', 'Manual'])
data, times = raw[:, :]
print(data)
print(times)
plt.plot(times, data.T)
plt.title("EEG C3-A2")
plt.show()
```

larsoner, 2022-10-21:
This appears to be a question about usage, not a bug report or a new feature request. Please use the [MNE mailing list](http://mail.nmr.mgh.harvard.edu/mailman/listinfo/mne_analysis) or the [gitter channel](https://gitter.im/mne-tools/mne-python) for usage questions.

larsoner, 2022-10-21:
Or actually https://mne.discourse.group/ is the best option nowadays!

closed 2022-10-21 by larsoner