mne-tools/mne-python #11274: Selection of ICA components when plotting properties

mscheltienne, 2022-10-24:
### Description of the problem

https://github.com/mne-tools/mne-python/blob/45d87fb0e5c34156c9568e04cd6ceef81140bed4/mne/viz/ica.py#L422-L423

Shouldn't this be `limit=min(5, ica.n_components_) if picks is None else ica.n_components_`?
Consider an ICA fitted with 55 components on 57 EEG channels:

```
ica.plot_properties(raw, picks=55)  # max is 54 since it picks on components
```

will currently fail with `IndexError: index 55 is out of bounds for axis 1 with size 55`, a bit after the previous code snippet, here: https://github.com/mne-tools/mne-python/blob/45d87fb0e5c34156c9568e04cd6ceef81140bed4/mne/viz/ica.py#L459

Overall, I think the `limit` should be updated and the validity of the provided `picks` should be asserted at the beginning of this function, with an error message more informative. 

### Steps to reproduce

```Python
-
```

### Expected results

-

### Actual results

-

### Additional information

-

mscheltienne, 2022-10-27:
Before opening a PR, I rapidly checked the code that uses `_fast_plot_ica_properties`:
- `ICA.plot_ica_properties`, which is also used by `ica.plot_components(inst=...)`, as expected
- in `mne.viz._figure.py`, in `BrowserBase._create_ica_properties_fig` 

https://github.com/mne-tools/mne-python/blob/cee4e7d11116bd4e0e219fe611c0121cafd2394d/mne/viz/_figure.py#L475-L498

I'm a bit concern by this second one and by what type might be stored in `self.mne.ica._ica_names`.. `BrowserBase._create_ica_properties_fig` is called in `_create_ch_context_fig`:

https://github.com/mne-tools/mne-python/blob/cee4e7d11116bd4e0e219fe611c0121cafd2394d/mne/viz/_figure.py#L437-L448

which is called by the matplotlib pick event in `MNEBrowseFigure`.

I don't get in which context this chain of functions is called, the only browser I can think of related to the ICA is `ica.plot_sources(inst)`, and I don't get where the `plot_properties` comes up in this one. I'm not very familiar with the plotting API, any idea?

larsoner, 2022-10-31:
I didn't want to have to follow our code so I tried trusting our tests by adding a `raise RuntimeError` early in `_fast_plot_ica_properties`, it looks like both failure modes are thankfully visible:
```
__________________________________________________________________________ test_plot_ica_properties __________________________________________________________________________
mne/viz/tests/test_ica.py:144: in test_plot_ica_properties
    ica.plot_properties(raw, picks=0, verbose='debug', **topoargs)
mne/preprocessing/ica.py:2205: in plot_properties
    return plot_ica_properties(self, inst, picks=picks, axes=axes,
<decorator-gen-164>:10: in plot_ica_properties
    ???
mne/viz/ica.py:392: in plot_ica_properties
    return _fast_plot_ica_properties(ica, inst, picks=picks, axes=axes, dB=dB,
mne/viz/ica.py:410: in _fast_plot_ica_properties
    raise RuntimeError
E   RuntimeError
_____________________________________________________________________ test_plot_ica_sources[matplotlib] ______________________________________________________________________
mne/viz/tests/test_ica.py:279: in test_plot_ica_sources
    fig._click_ch_name(ch_index=0, button=3)
mne/viz/_mpl_figure.py:2005: in _click_ch_name
    _click_ch_name(self, ch_index, button)
mne/utils/_testing.py:435: in _click_ch_name
    _fake_click(fig, fig.mne.ax_main, (x, y), xform='pix',
mne/viz/utils.py:788: in _fake_click
    fig.canvas.callbacks.process(
../matplotlib/lib/matplotlib/cbook/__init__.py:307: in process
    func(*args, **kwargs)
../matplotlib/lib/matplotlib/figure.py:2506: in pick
    super().pick(mouseevent)
../matplotlib/lib/matplotlib/artist.py:525: in pick
    a.pick(mouseevent)
../matplotlib/lib/matplotlib/artist.py:525: in pick
    a.pick(mouseevent)
../matplotlib/lib/matplotlib/artist.py:525: in pick
    a.pick(mouseevent)
../matplotlib/lib/matplotlib/artist.py:525: in pick
    a.pick(mouseevent)
../matplotlib/lib/matplotlib/artist.py:511: in pick
    mouseevent, self, **prop)._process()
../matplotlib/lib/matplotlib/backend_bases.py:1231: in _process
    self.canvas.callbacks.process(self.name, self)
../matplotlib/lib/matplotlib/cbook/__init__.py:307: in process
    func(*args, **kwargs)
mne/viz/_mpl_figure.py:779: in _pick
    self._create_ch_context_fig(ind)
mne/viz/_mpl_figure.py:782: in _create_ch_context_fig
    fig = super()._create_ch_context_fig(idx)
mne/viz/_figure.py:444: in _create_ch_context_fig
    fig = self._create_ica_properties_fig(pick)
mne/viz/_figure.py:494: in _create_ica_properties_fig
    _fast_plot_ica_properties(
mne/viz/ica.py:410: in _fast_plot_ica_properties
    raise RuntimeError
E   RuntimeError
```

drammock, 2022-11-01:
> I didn't want to have to follow our code

*aside* if even @larsoner feels this way, it seems we are ripe for a refactor / code readability improvements (var names, comments, etc) 😆

mscheltienne, 2022-11-01:
> aside if even @larsoner feels this way, it seems we are ripe for a refactor / code readability improvements (var names, comments, etc) 😆

That's a big tell ahah

Thank you @larsoner, I did not have time to look at it yet, but that's a great starting point! I'll try to dig in this week or next week.

PR #11369 (2022-12-15): Improve ICA component selection with the argument picks
Closes #11274 

This PR improves the selection of IC with the argument `picks`, especially making sure that the type and value provided are valid. 
- Adds an helper function `_check_pick_components` in `utils/check.py` (do you want it elsewhere?)
- Replace `picks = _picks_to_idx(ica.info, picks)` (which picks on the channel names!) with `_check_pick_components(ica, picks)`
- Improve the error message, c.f. #11274
- Fix the docstring for `picks` in `ica.plot_properties`.

In #11274, I wasn't sure in which context a Browser figure would ope [... cut]
closed 2022-12-15 by larsoner