mne-tools/mne-python #11286: [ENH, MAINT, BUG] Refactor mne.viz.Brain

alexrockhill, 2022-10-28:
### Describe the new feature or enhancement

To me `mne.viz.Brain` is a misnomer when there are multiple views; `matplotlib.pyplot.Axes` isn't a single subplot *or* multiple subplots, it's just a single subplot axis, I think `mne.viz.Brain` should just be a single 3D rendered brain object/scene.

### Describe your proposed implementation

The GUI code could all be pulled out into an `mne.viz.BrainViewer` or such that handles one or multiple `mne.viz.Brains` and synchronizes them.

### Describe possible alternatives

If the sidebar and the toolbar work synchronously for multiple views, it seems to me that those could only make sense at level of the `mne.viz.BrainViewer` if this approach were to be taken so that would need to be moved up a level as well.

### Additional context

I don't think this would be years or work, I think this would be a lot like abstracting `SliceBrowser` out of the `IEEGLocatorGUI`, a week or so.

larsoner, 2022-10-28:
Please read my posts from before a bit more carefully -- this is not what I suggested. I think we want to keep the Brain API the same (multiple views, methods, etc.) and just change the guts behind it.

alexrockhill, 2022-10-28:
I read your posts carefully and I agree on moving to the QInteractor and using the abstraction, those both seem like great ideas. This is separate but related and given your comment, I think it's clear I disagree with you that this is not a problem, I think it really is not the best way to do things and should be fixed.

larsoner, 2022-10-28:
I would much rather keep the API of Brain more or less as it is, and instead have what you want to make "Brain" be called "BrainView" or something that implies "one single view of the brain". The code churn / deprecation for people is not worth what more or less amounts to a slightly clearer name at our end. I think our threshold for deprecation should be higher than this

alexrockhill, 2022-11-03:
Ok, I'll go ahead and close the issue. I still think this is a design issue and since `mne.viz.Brain` is tagged as experimental it would be a good time to fix it but I'm not going to press the issue. I think in the long run, it will be something that really simplifies `mne.viz.Brain` but it works okay mostly now so not the highest priority.

closed 2022-11-03 by alexrockhill