mne-tools/mne-python #11288: add_volume_labels() method from mne.viz.get_brain_class() doesn't allow receiving a dict as legend

mcespedes99, 2022-10-30:
### Description of the problem

Hi,

In the MNE documentation for the add_volume_labels() method  from mne.viz.get_brain_class(), it's stated that you pass a dictionary to the legend option, which is actually implemented on the code. However, line 2475 from the _brain.py file actually does a wrong validation since it is set to `_validate_type(legend, (bool, None), 'legend')`; therefore, it actually doesn't allow using the dict option. It should be set to `validate_type(legend, (bool, None, dict), 'legend')` in order to make it work.

Best,
Mauricio

### Steps to reproduce

```Python
This is the code I was using to reproduce:
%matplotlib inline
import os.path as op
import mne
import numpy as np

sample_data_folder = mne.datasets.sample.data_path()
subjects_dir = sample_data_folder / 'subjects'

fname_aseg = op.join(subjects_dir, 'sample', 'mri', 'aseg.mgz')
label_names = mne.get_volume_labels_from_aseg(fname_aseg)

Brain2 = mne.viz.get_brain_class()
brain2 = Brain2('sample', hemi='both', surf='pial', alpha=0.2,
              cortex='low_contrast', subjects_dir=subjects_dir, 
              size=(800, 600))
plotargs = {
    'bcolor':(0.5, 0.5, 0.5), 
    'border':False,
    'size':(0.2, 0.6),
    'loc':'upper left'
}
brain2.add_volume_labels(aseg='aseg', labels=label_names, alpha=1, smooth=0, legend=plotargs)
```

### Expected results

The legend is shown in the qt widget.

### Actual results

Legend is not shown and an error is displayed.

### Additional information

Platform:         Linux-5.15.0-52-generic-x86_64-with-glibc2.35
Python:           3.10.6 | packaged by conda-forge | (main, Aug 22 2022, 20:35:26) [GCC 10.4.0]
Executable:       /home/mcespedes/miniconda3/envs/mne/bin/python
CPU:              x86_64: 8 cores
Memory:           15.4 GB

mne:              1.2.1
numpy:            1.23.4 {OpenBLAS 0.3.21 with 8 threads}
scipy:            1.9.3
matplotlib:       3.6.0 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.1.3
numba:            0.56.3
nibabel:          4.0.2
nilearn:          0.9.2
dipy:             1.5.0
openmeeg:         Not found
cupy:             Not found
pandas:           1.5.1
pyvista:          0.36.1 {OpenGL 4.6 (Core Profile) Mesa 22.0.5 via Mesa Intel(R) Xe Graphics (TGL GT2)}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
vtk:              9.2.2
qtpy:             2.2.1 {PyQt5=5.15.6}
ipympl:           Not found
...
mne_features:     Not found
mne_qt_browser:   0.4.0
mne_connectivity: Not found
mne_icalabel:     Not found

larsoner, 2022-10-31:
> It should be set to validate_type(legend, (bool, None, dict), 'legend') in order to make it work.

Have you tried making this change? If you do, does it work correctly? If so, a PR to fix this would be great!

mcespedes99, 2022-10-31:
Yes I have. Will create the PR then!

Thanks,
Mauricio

PR #11289 (2022-10-31): Update _brain.py to allow usage of dict in legends
Related to the issue #11288 

Thanks for contributing a pull request! Please make sure you have read the
[contribution guidelines](https://mne.tools/dev/install/contributing.html)
before submitting.

Please be aware that we are a loose team of volunteers so patience is
necessary. Assistance handling other issues is very welcome. We value
all user contributions, no matter how minor they are. If we are slow to
review, either the pull request needs some benchmarking, tinkering,
convincing, etc. or more likely the reviewers are simply busy. In eit [... cut]
closed 2022-10-31 by larsoner