mne-tools/mne-python #11294: scaling of beamformer images when no noise covariance is available

nugenta, 2022-11-02:
in `mne.beamformer._compute_beamformer.py`, in `_prepare_beamformer_input`, if no noise covariance matrix is used, then an adhoc covariance is created with this call:

`noise_cov = make_ad_hoc_cov(info, std=1.)`

Using an identity matrix for the noise covariance (instead of a diagonal matrix scaled by a noise estimate) results in incorrect scaling of the resultant beamformed images, depending on the choice of weight normalization. (i.e. using unit-noise-gain). 
I'm pretty sure that the call should instead be: `noise_cov = make_ad_hoc_cov(info, std=None)`, which uses canonical values for noise depending on sensor type. Obviously, this is an estimate, but when you use an identity matrix, the resulting source time series has values on the order of 1E-26, which isn't interpreted correctly when viewing `mne.viz.Brain`. I feel like using a somewhat realistic ad hoc covariance where you can actually use the sliders when visualizing the stc is preferable.

agramfort, 2022-11-02:
ping @britta-wstnr

larsoner, 2022-11-02:
Sounds reasonable to me, but happy to defer to @britta-wstnr here

britta-wstnr, 2022-11-03:
Mhhh ... @larsoner we still test against FieldTrip, don't we? That should have caught something going wrong?

larsoner, 2022-11-03:
Yes, but I guess it's a matter of interpretation/defaults. If they pass `noise_cov=None`, taking the gradiometer-only case for example, should we:

1. Assume that the noise level was 1 T/m (`main` and FieldTrip), or
2. Assume that the noise level was 5 fT/cm, as this is the default for `make_ad_hoc_cov` (what @nugenta proposes)

@nugenta while we hash this out, in the meantime you probably realized this already but you can pass `noise_cov=make_ad_hoc_cov(info)` to get your desired behavior already.

So here we "just" have to decide if our default makes sense or not...

larsoner, 2022-11-03:
Opened https://github.com/mne-tools/mne-python/pull/11302 so you can see what changes would be necessary @britta-wstnr 

cc @wmvanvliet since it also changes DICS a bit

nugenta, 2022-11-09:
Thanks everybody!

PR #11302 (2022-11-09): DOC: Suggest using make_ad_hoc_cov in beamformers
@britta-wstnr locally all tests pass with these changes. You can see what tests need to be adjusted... TBD on whether or not this is desired.

Closes #11294
closed 2022-11-09 by larsoner