mne-tools/mne-python #11295: Error in cross_val_multiscore

jiivana, 2022-11-02:
### Description of the problem

I am following the example on temporal decoding: https://mne.tools/stable/auto_tutorials/machine-learning/50_decoding.html#temporal-decoding to classify stimuli in three categories. The example runs on my Windows machine fine, but when I add PCA and/or CSP to the pipeline I get errors.

### Steps to reproduce

```Python
# We will train the classifier on all left visual vs auditory trials on MEG

csp = CSP(n_components=5, norm_trace=False)
clf = make_pipeline(StandardScaler(),
    UnsupervisedSpatialFilter(PCA(16), average=False),
    csp,
    LogisticRegression(solver='liblinear')
)

time_decod = SlidingEstimator(
    clf, n_jobs=1, verbose=False)

# here we use cv=3 just for speed
scores = cross_val_multiscore(time_decod, X, y, cv=3, n_jobs=1,verbose=False)

# Mean scores across cross-validation splits
scores = np.mean(scores, axis=0)
```

### Expected results

I expect the code to work..?

### Actual results

Traceback (most recent call last):

  File "XX\Python\Python39\site-packages\mne\decoding\search_light.py", line 80, in fit
    estimators = parallel(

  File "XX\\Python\Python39\site-packages\mne\decoding\search_light.py", line 81, in <genexpr>
    p_func(self.base_estimator, split, y, pb.subset(pb_idx),

  File "XX\\Python\Python39\site-packages\mne\decoding\search_light.py", line 318, in _sl_fit
    est.fit(X[..., ii], y, **fit_params)

  File "C:\ProgramData\Anaconda3\envs\mne\lib\site-packages\sklearn\pipeline.py", line 390, in fit
    Xt = self._fit(X, y, **fit_params_steps)

  File "C:\ProgramData\Anaconda3\envs\mne\lib\site-packages\sklearn\pipeline.py", line 348, in _fit
    X, fitted_transformer = fit_transform_one_cached(

  File "C:\ProgramData\Anaconda3\envs\mne\lib\site-packages\joblib\memory.py", line 349, in __call__
    return self.func(*args, **kwargs)

  File "C:\ProgramData\Anaconda3\envs\mne\lib\site-packages\sklearn\pipeline.py", line 893, in _fit_transform_one
    res = transformer.fit_transform(X, y, **fit_params)

  File "XX\Python\Python39\site-packages\mne\decoding\transformer.py", line 628, in fit_transform
    return self.fit(X).transform(X)

  File "XX\Python\Python39\site-packages\mne\decoding\transformer.py", line 606, in fit
    n_epochs, n_channels, n_times = X.shape

ValueError: not enough values to unpack (expected 3, got 2)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "XX", line 99, in <module>
    scores = cross_val_multiscore(time_decod, X, y, cv=3, n_jobs=1,verbose=False)

  File "<decorator-gen-469>", line 10, in cross_val_multiscore

  File "XX\Python\Python39\site-packages\mne\decoding\base.py", line 435, in cross_val_multiscore
    scores = parallel(

  File "XX\Python\Python39\site-packages\mne\decoding\base.py", line 436, in <genexpr>
    p_func(

  File "XX\Python\Python39\site-packages\mne\decoding\base.py", line 470, in _fit_and_score
    estimator.fit(X_train, y_train, **fit_params)

  File "XX\Python\Python39\site-packages\mne\decoding\search_light.py", line 80, in fit
    estimators = parallel(

  File "X\Python\Python39\site-packages\mne\utils\progressbar.py", line 157, in __exit__
    os.remove(self._mmap_fname)

PermissionError: [WinError 32] The process cannot access the file because it is being used by another process: 'XX\AppData\\Local\\Temp\\tmp_mne_prog6xxe5_fn'

### Additional information

Platform:         Windows-10-10.0.19044-SP0
Python:           3.9.7 | packaged by conda-forge | (default, Sep 29 2021, 19:15:42) [MSC v.1916 64 bit (AMD64)]
Executable:       C:\ProgramData\Anaconda3\envs\mne\python.exe
CPU:              Intel64 Family 6 Model 158 Stepping 13, GenuineIntel: 12 cores
Memory:           15.8 GB

mne:              1.2.1
numpy:            1.21.4 {unknown linalg bindings}
scipy:            1.7.2
matplotlib:       3.4.3 {backend=Qt5Agg}

sklearn:          1.0.1
numba:            0.53.1
nibabel:          3.2.1
nilearn:          0.8.1
dipy:             1.4.1
openmeeg:         Not found
cupy:             Not found
pandas:           1.3.4
pyvista:          0.32.1 {OpenGL 4.5.0 NVIDIA 452.96 via Quadro T2000/PCIe/SSE2}
pyvistaqt:        0.5.0
ipyvtklink:       Not found
vtk:              9.0.3
qtpy:             1.11.2 {PyQt5=5.12.9}
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.5.2

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

drammock, 2022-11-02:
looks like the error comes from within `UnsupervisedSpatialFilter.fit()`

I'm not super familar with this kind of analysis, but I think the problem is that `UnsupervisedSpatialFilter` expects there to be a time dimension, but when you wrap it in a pipeline passed to `SlidingEstimator` then it no longer gets access to the time dimension?  IIUC, `SlidingEstimator` "consumes" the time dimension and passes each timepoint into the pipeline separately.

(take that with a grain of salt; as I said I'm not super familiar with such analyses. @agramfort might have better insight)

larsoner, 2022-11-03:
That sounds right @drammock

We could maybe make `UnsupervisedSpatialFilter` deal with a lack of "time" dimension by treating "epochs" as times basically. Really we just need some repeated measures across sensors -- be it across the time dim or epochs dim -- so this could in principle work.

But one probably better possibility is to create the opposite of https://mne.tools/stable/generated/mne.decoding.Vectorizer.html that will promote a 2D `(n_samples, n_features)` array to a `(n_samples // n_times, n_features, n_times)` array, e.g. `mne.decoding.Unvectorizer(n_times=1)` for your case would promote it to an array of shape `(n_epochs, n_sensors, 1)` which could then be used in `UnsupervisedSpatialFilter.fit()`. This idea is half-baked but it maybe could work...

agramfort, 2022-11-03:
CSP works by taking time samples as input to produce covariances and then obtain spatial filters. If you use a sliding estimator you work with one time point. So I am not sure of what you want to do. Besices CSP if for induced effects while SlidingEstimator captures time locked effects.

larsoner, 2022-11-03:
> CSP works by taking time samples as input to produce covariances and then obtain spatial filters.

But it does so by combining *time points across all epochs being fit*, right? So if you have for example 100 epochs and a 5-fold CV, you fit using 80 epochs and 1 time point apiece -> 80 samples to compute the covariance for your 16 channels. So I think the `Unvectorizer` approach could potentially work in this case if you have enough epochs, right?

agramfort, 2022-11-03:
hum ok that could work on paper...

> Message ID: ***@***.***>
>

larsoner, 2022-11-04:
@jiivana do you think this would work for your use case if it were implemented?

drammock, 2022-11-04:
So the summary is

```py
SlidingEstimator(make_pipeline(..., UnsupervisedSpatialFilter(...), ...), ...)
```

doesn't work (and presumably also won't work with `CSP` in place of / in addition to `UnsupervisedSpatialFilter`). But if we provided an `Unvectorizer` then that approach could still work in principle, by allowing `UnsupervisedSpatialFilter` or `CSP` to aggregate over *epochs* instead of over *time points*. Yes?

jiivana, 2022-11-04:
I should do my homework in understanding how these methods work. I think I tried to use `SlidingEstimator` to look at the csp performance over time similarly to what is done in https://mne.tools/stable/auto_examples/decoding/decoding_csp_eeg.html. So I think this is just confusion from my part.

drammock, 2022-11-04:
OK. Feel free to re-open if you find that you can't accomplish your analysis goal.

closed 2022-11-04 by drammock