mne-tools/mne-python #11299: find_bads_muscle() can't find sensor positions

skjerns, 2022-11-03:
### Description of the problem

When using the newly introduced `ica.find_bads_muscle` it seems like the function cannot find the sensor positions, even though they should be included in the file and/or be accessible to MNE generally by default.

```Python
ValueError: The following electrodes have overlapping positions, which causes problems during visualization: [... list of all 306 channels]
```

### Steps to reproduce

```Python
import mne
from mne.preprocessing import ICA

raw = mne.io.read_raw('testfile.fif', preload=True)

ica =ICA(40, method='picard')
ica.fit(raw)

ica.find_bads_muscle(raw)
```

### Link to data

Using this sample file from our Trilux/MEGIN system: 
https://cloud.skjerns.de/index.php/s/MqFecWkX3KExQPA

### Expected results

x

### Actual results

```Python
ValueError: The following electrodes have overlapping positions, which causes problems during visualization: [... list of all 306 channels]
```

### Additional information

x

mscheltienne, 2022-11-03:
Hello, I could reproduce the issue and I've pushed a fix in #11300.

skjerns, 2022-11-03:
Thanks for the fast fix!

However, I'm a bit surprised. Using your fix `ICA.find_bads_muscle` finds zero components in all of my files. My dataset of 30 participants with each 7 different recordings (rangin from 5-30 minutes), recorded with a MEGIN-Trilux with 306 sensors, no component gets marked as belonging to muscle artefacts. Is that to be expected (as MEG suffers less from muscle artefacts)?

drammock, 2022-11-03:
if they're adults, they're reasonably good at sitting still, and you allowed them to move in between trials/blocks to make it easier to be still during the blocks, it doesn't seem that surprising to me.

PR #11300 (2022-11-03): [BUG] Fix channel selection when retrieving positions in ICA.find_bads_muscle
Closes #11299 
cc @alexrockhill since you authored this function in #10534.
closed 2022-11-03 by drammock