mne-tools/mne-python #11314: Pick_Channels Returns AttributeError: 'NoneType' object has no attribute 'items

aroyphillips, 2022-11-10:
### Description of the problem

Hello,

I am trying to select a subset of the channels in my Raw instance using the `pick_channels` function. However,  it appears that the Raw instance has orig_units set to None, causing an error. Despite this, I'm actually still able to select the subset as the error seems to occur after the issue has happened. I'm able to get around this by using `try... except`, but I wanted to alert the team of this possible error.

For reference, this raw instance was created by reading from NKT file, accessing the data and filtering it directly, and then making a new raw file with the filtered data. Here is what the workflow looks like:
```
# load data from test_file
raw = mne.io.read_raw_nihon(test_file, preload=True

# filter data
arry = raw.get_data()
filtered_arry = utils.filter_eeg(arry, fs=fs, l_freq=l_freq, h_freq=fs/2-h_freq_cap, notches=fnoises).T # custom sos filter
new_raw = mne.io.RawArray(filtered_arry, raw.info)
new_raw.set_annotations(raw.annotations)

# select subset of channels
eeg_channels = ['Fp1', 'Fp2', 'F3', 'F4', 'C3', 'C4', 'P3', 'P4', 'O1', 'O2', 'F7', 'F8', 'T3', 'T4', 'T5', 'T6', 'Fz', 'Cz', 'Pz', 'T1', 'T2']
raw_copy = new_raw.copy()
raw_copy.pick_channels(eeg_channels)
```

Error message: 
```
AttributeError                            Traceback (most recent call last)
/home/roy/Desktop/github/mTBI_FITBIR/notebooks/feature_engineering_scratchwork.ipynb Cell 18 in <cell line: 2>()
      [1](vscode-notebook-cell://ssh-remote%2Bdestroyer.ece.rice.edu/home/roy/Desktop/github/mTBI_FITBIR/notebooks/feature_engineering_scratchwork.ipynb#X22sdnNjb2RlLXJlbW90ZQ%3D%3D?line=0) eeg_channels = ['Fp1', 'Fp2', 'F3', 'F4', 'C3', 'C4', 'P3', 'P4', 'O1', 'O2', 'F7', 'F8', 'T3', 'T4', 'T5', 'T6', 'Fz', 'Cz', 'Pz', 'T1', 'T2']
----> [2](vscode-notebook-cell://ssh-remote%2Bdestroyer.ece.rice.edu/home/roy/Desktop/github/mTBI_FITBIR/notebooks/feature_engineering_scratchwork.ipynb#X22sdnNjb2RlLXJlbW90ZQ%3D%3D?line=1) raw_copy.pick_channels(eeg_channels)

File <decorator-gen-47>:12, in pick_channels(self, ch_names, ordered, verbose)

File ~/anaconda3/envs/jupyterEnv/lib/python3.10/site-packages/mne/channels/channels.py:682, in UpdateChannelsMixin.pick_channels(self, ch_names, ordered, verbose)
    647 """Pick some channels.
    648 
    649 Parameters
   (...)
    679 .. versionadded:: 0.9.0
    680 """
    681 picks = pick_channels(self.info['ch_names'], ch_names, ordered=ordered)
--> 682 return self._pick_drop_channels(picks)

File <decorator-gen-49>:12, in _pick_drop_channels(self, idx, verbose)

File ~/anaconda3/envs/jupyterEnv/lib/python3.10/site-packages/mne/channels/channels.py:834, in UpdateChannelsMixin._pick_drop_channels(self, idx, verbose)
    831 if isinstance(self, BaseRaw):
    832     self.annotations._prune_ch_names(self.info, on_missing='ignore')
    833     self._orig_units = {
--> 834         k: v for k, v in self._orig_units.items()
...
    835         if k in self.ch_names}
    837 self._pick_projs()
    838 return self

AttributeError: 'NoneType' object has no attribute 'items'
```

### Steps to reproduce

```Python
# load data from test_file
raw = mne.io.read_raw_nihon(test_file, preload=True

# filter data
arry = raw.get_data()
filtered_arry = utils.filter_eeg(arry, fs=fs, l_freq=l_freq, h_freq=fs/2-h_freq_cap, notches=fnoises).T # custom sos filter
new_raw = mne.io.RawArray(filtered_arry, raw.info)
new_raw.set_annotations(raw.annotations)

# select subset of channels
eeg_channels = ['Fp1', 'Fp2', 'F3', 'F4', 'C3', 'C4', 'P3', 'P4', 'O1', 'O2', 'F7', 'F8', 'T3', 'T4', 'T5', 'T6', 'Fz', 'Cz', 'Pz', 'T1', 'T2']
raw_copy = new_raw.copy()
raw_copy.pick_channels(eeg_channels)
```
```

### Expected results

Should just run without error

### Actual results

AttributeError: 'NoneType' object has no attribute 'items'

### Additional information

Output exceeds the [size limit](command:workbench.action.openSettings?%5B%22notebook.output.textLineLimit%22%5D). Open the full output data[ in a text editor](command:workbench.action.openLargeOutput?2463ca9b-cc87-428b-888e-90d4e994d2b9)
Platform:         Linux-5.15.0-48-generic-x86_64-with-glibc2.35
Python:           3.10.4 | packaged by conda-forge | (main, Mar 24 2022, 17:38:57) [GCC 10.3.0]
Executable:       /home/roy/anaconda3/envs/jupyterEnv/bin/python
CPU:              x86_64: 64 cores
Memory:           1006.6 GB

mne:              1.2.1
numpy:            1.21.1 {OpenBLAS 0.3.20 with 64 threads}
scipy:            1.8.1
matplotlib:       3.5.2 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.1.1
numba:            0.55.1
nibabel:          3.2.2
nilearn:          0.9.1
dipy:             1.5.0
openmeeg:         Not found
cupy:             Not found
pandas:           1.4.4
pyvista:          0.34.0 {OpenGL could not be initialized}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
vtk:              9.1.0
qtpy:             2.2.0 {PyQt5=5.15.3}
ipympl:           0.9.2
...
mne_features:     Not found
mne_qt_browser:   0.3.1
mne_connectivity: Not found
mne_icalabel:     Not found

mscheltienne, 2022-11-10:
Hello, thank you for reporting the issue.
Could you share a file that raises this error, otherwise it will be difficult to reproduce?

Also, could you try in an environment running the latest development version. You can install it with:

```
$ pip install git+https://github.com/mne-tools/mne-python
```

I think this bug was already fixed in #11160

mscheltienne, 2022-11-10:
Or running the latest stable version, 1.2.0 or 1.2.1, the fix should be released already.

aroyphillips, 2022-11-11:
Hi, thank you for the response.

My mne version is up to date at 1.2.1. However, while trying to create a simple code to replicate the issue, I figured out the issue seems to be due to the pickle saving protocol. Saving the files as `.fif` format seems to fix it, even after loading with pickle, but since I'm working with a lot of raw files I have been opting to save entire lists of raw files at once with pickle.

Here is my code that reproduces the error. I'm not sure how to upload my data file, but I suspect that saving any raw file with pickle could yield the same error.
```
import mne
import pickle
# load data 
sample_file = '/home/roy/Desktop/github/mTBI_FITBIR/sample_data/raw_sample.pickle'
with open(sample_file, 'rb') as f:
    raw_sample = pickle.load(f)
# select subset of channels
eeg_channels = ['Fp1', 'Fp2', 'F3', 'F4', 'C3', 'C4', 'P3', 'P4', 'O1', 'O2', 'F7', 'F8', 'T3', 'T4', 'T5', 'T6', 'Fz', 'Cz', 'Pz', 'T1', 'T2']
raw_copy = raw_sample.copy()
print(raw_copy.ch_names)
raw_copy.pick_channels(eeg_channels)
```

mscheltienne, 2022-11-11:
Oh, most likely yes and someone might be able to pinpoint what went wrong, but that would take me too long to figure out. Pickle is not a long-lasting format, as such it is not supported by MNE and I can't recommend using it.

If you have many recordings to work with, you should define a clear naming schema that can be programmatically generated. For instance, if you have multiple subjects, multiple sessions and multiple runs per session, you could define a file name as:

```
sub_idx = 123
ses_idx = 1
run_idx = 5
f"sub-{str(sub_idx).zfill(3)}_ses-{str(ses_idx).zfill(2)}_run-{str(run_idx).zfill(2)}-raw.fif"
# >>> "sub-123_ses-01_run-05-raw.fif"
```

Or you could use the BIDS standard and `mne-bids` which will handle the file location/name.

closed 2026-08-18 by larsoner