mne-tools/mne-python #11316: Missing frame of head

DarthSaddest, 2022-11-10:
### Description of the problem

Steps to replicate:
* Read in the raw file from the [**sub-01/ses-iemu/ieeg/** directory](https://openneuro.org/datasets/ds003688/versions/1.0.7) 
* Remove bad channels and non-EEG channels with aid of `channels.tsv`.
* Load the montage from `electrodes.tsv` into `raw`. A warning is displayed: `RuntimeWarning: Fiducial point nasion not found, assuming identity unknown to head transformation`.
* On querying `raw.get_montage().get_positions()`, the output is:`{'ch_pos': OrderedDict([('AR1', array([ 22.42,  23.64, -31.74])),
              ('AR2', array([ 25.334,  23.354, -31.654])),
              ('AR3', array([ 31.334,  22.954, -32.054])),
              ('AR4', array([ 37.134,  22.354, -32.454])),
              ('AR5', array([ 42.934,  21.954, -32.854])),
              ('AR6', array([ 48.968,  21.521, -33.121])),
              ('AR7', array([ 54.734,  21.154, -33.454])),
              ('AR9', array([ 66.668,  20.621, -34.788])),
              ('AHR1', array([ 26.134,  15.354, -36.454])),
              ('AHR2', array([ 28.134,  15.354, -36.454])),
              ('AHR3', array([ 34.634,  14.354, -34.954])),
              ('AHR4', array([ 40.356,  13.465, -33.676])),
              ('AHR5', array([ 46.225,  12.627, -32.273])),
              ('AHR6', array([ 51.801,  11.799, -30.899])),
              ('AHR7', array([ 57.468,  10.91 , -30.01 ])),
              ('AHR8', array([ 63.579,  10.243, -28.788])),
              ('AHR9', array([ 69.134,   9.354, -27.454])),
              ('PHR1', array([ 28.706,  -5.646, -24.597])),
              ('PHR2', array([ 31.634,  -6.146, -24.954])),
              ('PHR3', array([ 37.734,  -5.846, -25.454])),
              ('PHR4', array([ 43.468,  -6.146, -26.121])),
              ('PHR5', array([ 49.634,  -5.646, -26.454])),
              ('PHR6', array([ 55.468,  -5.812, -26.621])),
              ('PHR7', array([ 61.634,  -5.646, -26.454])),
              ('PHR8', array([ 67.596,  -6.261, -26.07 ])),
              ('PHR9', array([ 73.075,  -6.587, -25.984])),
              ('ITR1', array([ 40.968,   5.521, -40.454])),
              ('ITR2', array([ 44.134,   5.354, -39.454])),
              ('ITR3', array([ 48.134,   4.354, -37.454])),
              ('ITR4', array([ 51.134,   3.354, -36.454])),
              ('ITR5', array([ 54.134,   3.354, -34.454])),
              ('ITR6', array([ 57.134,   3.354, -33.454])),
              ('ITR7', array([ 60.134,   2.354, -31.454])),
              ('ITR8', array([ 63.134,   2.354, -30.454])),
              ('ITR9', array([ 67.134,   2.354, -28.454])),
              ('ITR10', array([ 70.134,   1.354, -26.454])),
              ('MFR1', array([ 13.134,  70.354, -15.454])),
              ('MFR2', array([ 14.134,  70.354, -11.454])),
              ('MFR3', array([15.134, 70.354, -8.454])),
              ('MFR4', array([16.134, 70.354, -5.454])),
              ('MFR5', array([18.134, 70.354, -2.454])),
              ('MFR6', array([19.134, 70.354,  1.546])),
              ('MFR7', array([20.134, 69.354,  4.546])),
              ('MFR8', array([22.134, 69.354,  8.546])),
              ('MFR9', array([23.134, 69.354, 11.546])),
              ('MFR10', array([24.134, 69.354, 14.546])),
              ('MFR11', array([25.134, 69.354, 17.546])),
              ('MFR12', array([26.134, 69.354, 20.546])),
              ('MFR13', array([27.134, 69.354, 23.546])),
              ('MFR14', array([28.134, 68.354, 27.546])),
              ('MFR15', array([29.134, 68.354, 30.546])),
              ('FOR1', array([ 22.134,  67.354, -20.454])),
              ('FOR2', array([ 22.134,  66.354, -17.454])),
              ('FOR3', array([ 22.134,  65.354, -13.454])),
              ('FOR4', array([23.134, 65.354, -9.454])),
              ('FOR5', array([24.134, 65.354, -6.454])),
              ('FOR6', array([25.134, 65.354, -2.454])),
              ('FOR7', array([25.134, 64.354,  0.546])),
              ('FOR8', array([26.134, 64.354,  4.546])),
              ('FOR9', array([26.134, 64.354,  7.546])),
              ('FOR10', array([27.134, 64.354, 11.546])),
              ('FOR11', array([27.134, 63.354, 14.546])),
              ('FOR12', array([28.134, 63.354, 18.546])),
              ('FOR13', array([28.134, 63.354, 21.546])),
              ('FOR14', array([29.134, 63.354, 24.546])),
              ('FOR15', array([29.134, 62.354, 28.546])),
              ('MFL1', array([  6.134,  75.354, -15.454])),
              ('MFL2', array([  6.134,  75.354, -12.454])),
              ('MFL3', array([ 5.134, 75.354, -8.454])),
              ('MFL4', array([ 4.134, 75.354, -4.454])),
              ('MFL5', array([ 2.134, 74.354,  1.546])),
              ('MFL6', array([ 1.134, 74.354,  4.546])),
              ('MFL7', array([ 1.134, 74.354,  7.546])),
              ('MFL8', array([ 0.134, 73.354, 11.546])),
              ('MFL9', array([-0.866, 73.354, 15.546])),
              ('MFL10', array([-1.866, 73.354, 18.546])),
              ('MFL11', array([-1.866, 72.354, 22.546])),
              ('MFL12', array([-2.866, 72.354, 24.546])),
              ('MFL13', array([-3.866, 72.354, 28.546])),
              ('MFL14', array([-4.866, 71.354, 32.546])),
              ('ACR1', array([16.134, 56.354, -8.454])),
              ('ACR2', array([16.134, 56.354, -5.454])),
              ('ACR3', array([17.134, 56.354, -2.454])),
              ('ACR4', array([18.134, 56.354,  1.546])),
              ('ACR5', array([19.134, 56.354,  5.546])),
              ('ACR6', array([19.134, 56.354,  9.546])),
              ('ACR7', array([20.134, 56.354, 11.546])),
              ('ACR8', array([21.134, 56.354, 15.546])),
              ('ACR9', array([21.134, 56.354, 18.546])),
              ('ACR10', array([22.134, 56.354, 22.546])),
              ('ACR11', array([22.134, 56.354, 25.546])),
              ('ACR12', array([23.134, 56.354, 29.546])),
              ('AHL1', array([-22.066,  14.354, -32.854])),
              ('AHL2', array([-24.866,  15.354, -32.621])),
              ('AHL3', array([-30.58 ,  17.212, -32.883])),
              ('AHL4', array([-36.466,  18.554, -32.454])),
              ('AHL5', array([-42.199,  20.188, -32.621])),
              ('AHL6', array([-47.699,  22.354, -32.454])),
              ('AHL7', array([-53.366,  23.854, -31.954])),
              ('AHL8', array([-59.366,  25.354, -31.954])),
              ('AHL9', array([-65.097,  26.97 , -31.762]))]),
 'coord_frame': 'head',
 'nasion': array([ 0.       , 66.3410594,  0.       ]),
 'lpa': array([-66.3410594,   0.       ,   0.       ]),
 'rpa': array([66.3410594,  0.       ,  0.       ]),
 'hsp': None,
 'hpi': None}`.
* `raw.plot_psd(fmax=100)` generates a multi-color plot corresponding to the electrodes. At the top left is the topo-legend but the frame of the head is missing!

### Steps to reproduce

```Python
Already provided.
```

### Expected results

The head-frame ought be visisble!

### Actual results

[Screenshot](https://ibb.co/y6Kwmhm)

### Additional information

Platform:         Windows-10-10.0.22000-SP0
Python:           3.9.7 (tags/v3.9.7:1016ef3, Aug 30 2021, 20:19:38) [MSC v.1929 64 bit (AMD64)]
Executable:       C:\Users\DarthSaddest\AppData\Local\Programs\Python\Python39\python.exe
CPU:              Intel64 Family 6 Model 140 Stepping 1, GenuineIntel: 8 cores
Memory:           7.7 GB
mne:              1.2.1
numpy:            1.22.3 {unknown linalg bindings (threadpoolctl module not found: No module named 'threadpoolctl')}
scipy:            1.9.3
matplotlib:       3.6.2 {backend=module://matplotlib_inline.backend_inline}
sklearn:          Not found
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           1.5.1
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0
mne_bids:         0.11.1
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

drammock, 2022-11-10:
I suspect that the head is there, but it's drawn really really small (because it thinks the sensors are several *meters* apart instead of several *millimeters* apart.  Please double-check that the units of your sensor positions are correct.  MNE-Python requires that they be in meters; if they were stored as millimeters then their locations will be off by a factor of 1000.

drammock, 2022-11-10:
...in fact, looking more closely at what you pasted in: `'ch_pos': OrderedDict([('AR1', array([ 22.42, 23.64, -31.74]))...` those numbers 22.42, 23.64, 31.74 are being interpreted as meters. And in the dataset they are the same: https://openneuro.org/datasets/ds003688/versions/1.0.7/file-display/sub-01:ses-iemu:ieeg:sub-01_ses-iemu_acq-clinical_electrodes.tsv

cc @sappelhoff is the measurement unit of the electrode locations stored anywhere in a BIDS dataset?  It doesn't seem to be in the above `electrodes.tsv`...  Or is it mandated to always be in m or cm or mm?

DarthSaddest, 2022-11-10:
It appears that Francois Tadel came across a similar (?) problem while using this dataset (scroll to the bottom):

> I'm testing your dataset with BIDS importer of Brainstorm, and I'm having issues with the interpretation of the ACPC coordinates available in the _electrodes.tsv files. With the conversion from ACPC coordinates to MRI coordinates in Brainstorm, most of the contacts are too far out. [..] I'm not sure if the issue is due to a bug in the conversion code in Brainstorm, or if the coordinates are incorrect in this dataset.

drammock, 2022-11-10:
I checked the BIDS standard and it says measurement units are given in `*coordsystem.json`. Looking at [that file](https://openneuro.org/datasets/ds003688/versions/1.0.7/file-display/sub-01:ses-iemu:ieeg:sub-01_ses-iemu_acq-clinical_coordsystem.json) I see that it does specify millmeters, so either (1) you need to multiply the locations by `1e-3` before setting the montage, or (2) load the data with MNE-BIDS (e.g., https://mne.tools/mne-bids/stable/auto_examples/read_bids_datasets.html) --- I assume that MNE-BIDS will properly handle the coordsystem file and scale the electrode locations automatically.  If it doesn't, please open a new issue about it on the MNE-BIDS repo.

closed 2022-11-10 by drammock

DarthSaddest, 2022-11-10:
@drammock Thanks, reading the raw file **directly** via MNE-BIDS works!

DarthSaddest, 2022-11-11:
P.S: I cannot but wonder if mne-python shall fix their reading of montage to take coordsystem.json into account, maybe as an optional parameter?

drammock, 2022-11-15:
> P.S: I cannot but wonder if mne-python shall fix their reading of montage to take coordsystem.json into account, maybe as an optional parameter?

an optional parameter to what?  `make_dig_montage()`?  That seems like the most logical place to do it to me, but I think I'm still not in support of that change.  If a dataset is in BIDS format, users should use mne-bids to read it in.  If it's not in BIDS format, it probably won't have `coordsystem.json` anyway.

sappelhoff, 2022-11-16:
I agree with @drammock on this