mne-tools/mne-python #11319: Incorporate Moving Window Average for mne.time_frequency.EpochsTFR

GaretEdmund, 2022-11-11:
### Describe the new feature or enhancement

I wish MNE to incorporate a method of taking a moving window average of spectral power wrt time. 

A methodology of choice, used by multiple researchers, goes roughly:
> 1) Use the Morlet wavelet transform to find spectral power as a function of time.
> 2) Add a buffer on both sides of the data to prevent edge effects. 
> 3) Sample frequencies either linearly or logarithmically at finite intervals between a range. 
> 4) Down-sample power values by taking a moving average across small-time windows and slide the window at a particular rate.

Current MNE functions allow (1), (2), and (3) but not (4).

### Describe your proposed implementation

This averaging method can be added to the mne.time_frequency.EpochsTFR class.

### Describe possible alternatives

None.

agramfort, 2022-11-11:
Can you share a code snippet to do 1 to 3 so I can see what’s your approach ?

GaretEdmund, 2022-11-11:
Something like this excluding 2:
```
#F_min, F_max, x and y need to be initialised per requirements.
power = mne.time_frequency.tfr_morlet(epoch, freqs=np.logspace(*np.log10([F_min, F_max]), num=x), n_cycles=y, use_fft=True, return_itc=False)
```

agramfort, 2022-11-13:
can you provide a full code snippet eg using the mne sample dataset so
we can just run it and see what to add and with some inline comments ?

mmagnuski, 2022-11-19:
@GaretEdmund
I'm not familiar with the fourth step in the methodology you describe, can you share some reference? (I mean, I'm familiar with moving average, of course, but not in the context of postprocessing time-frequency results)
If you want to have a lower sampling rate in your time-frequency representation, you can use `decim` argument to decimate (subsample) the TFR. But if you still prefer to take a moving window average, I think using xarray would be handy.

Below is piece of code is a modified version of parts of [this example](https://mne.tools/dev/auto_examples/time_frequency/time_frequency_erds.html#sphx-glr-auto-examples-time-frequency-time-frequency-erds-py), it uses `mne.datasets.eegbci`.

```python

import numpy as np
import matplotlib.pyplot as plt

import mne
from mne.datasets import eegbci
from mne.io import concatenate_raws, read_raw_edf
from mne.time_frequency import tfr_multitaper

# %% read eegbci data and construct epochs
fnames = eegbci.load_data(subject=1, runs=(6, 10, 14))
raw = concatenate_raws([read_raw_edf(f, preload=True) for f in fnames])
raw.rename_channels(lambda x: x.strip('.'))  # remove dots from channel names
events, _ = mne.events_from_annotations(raw, event_id=dict(T1=2, T2=3))

tmin, tmax = -1, 4
event_ids = dict(hands=2, feet=3)  # map event IDs to tasks
epochs = mne.Epochs(raw, events, event_ids, tmin - 0.5, tmax + 0.5,
                    picks=('C3', 'Cz', 'C4'), baseline=None, preload=True)

# %% calculate TFR
winlen = 0.4
freq_bandwidth = 8
time_band_prod = freq_bandwidth * winlen
freqs = np.arange(2, 80)
baseline = (-1, 0.2)

tfr = tfr_multitaper(epochs, freqs=freqs, n_cycles=freqs * winlen, use_fft=True,
                     return_itc=False, average=False, decim=4).average()
tfr_avg = tfr.crop(tmin, tmax).apply_baseline(baseline, mode="percent")

# %% construct xarray out of tfr
# (could be made easier with `.to_xarray()` method as discussed here:
#  https://github.com/mne-tools/mne-python/issues/7793)
import xarray as xr

tfr_xr = xr.DataArray(tfr_avg.data, dims=['chan', 'freq', 'time'],
                      coords={'freq': tfr_avg.freqs, 'time': tfr_avg.times,
                              'chan': tfr_avg.ch_names})

# %% perform moving average
tfr_xr_smoothed = tfr_xr.rolling(time=5, center=True).mean()

# %% plot the TFR before vs after the averaging
fig, ax = plt.subplots(ncols=2, figsize=(10, 4))
tfr_xr.sel(chan='Cz').plot(ax=ax[0])
tfr_xr_smoothed.sel(chan='Cz').plot(ax=ax[1])
```

Bear in mind that you can also control the time-resolution by changing the number of cycles (`n_cycles`) of preferred TFR function. To offset increased frequency resolution (if you prefer to have some added frequency smoothing) that a longer time window brings you can use multitapering.