mne-tools/mne-python #11326: ValueError: invalid literal for int() with base 10: ''

jo780-full, 2022-11-14:
### Description of the problem

Hello, I am new to mne and have a problem with this dataset.

It shows 
ValueError: invalid literal for int() with base 10: ''

### Steps to reproduce

```Python
`datatype='eeg'
subject='024'
task='med1breath'
suffix='eeg'
bids_path=BIDSPath(subject=subject,task=task,suffix=suffix,datatype=datatype,root=bids_root)
sidecar_path=bids_path.copy().update(extension='.json')
raw = read_raw_bids(bids_path=bids_path)
`
```

### Link to data

[doi:10.18112/openneuro.ds003969.v1.0.0](https://doi.org/10.18112/openneuro.ds003969.v1.0.0)]

### Expected results

Reads file

### Actual results

[image]

### Additional information

Platform:         Windows-10-10.0.22000-SP0
Python:           3.9.7 (default, Sep 16 2021, 16:59:28) [MSC v.1916 64 bit (AMD64)]
Executable:       C:\Users\josep\anaconda3\python.exe
CPU:              AMD64 Family 23 Model 96 Stepping 1, AuthenticAMD: 16 cores
Memory:           15.4 GB

mne:              1.2.2
numpy:            1.20.3 {MKL 2021.4-Product with 8 threads}
scipy:            1.7.1
matplotlib:       3.4.3 {backend=module://matplotlib_inline.backend_inline}

sklearn:          0.24.2
numba:            0.54.1
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           1.3.4
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             1.10.0 {PyQt5=5.9.7}
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         0.11.1
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

mscheltienne, 2022-11-15:
Hello,

I can not reproduce. I downloaded subject 24 from this dataset using `openneuro-py`, and loaded it successfully with:

```
from mne_bids import BIDSPath, read_raw_bids

bids_root = "/Users/scheltie/Documents/datasets/openneuro/ds003969/"
bids_path = BIDSPath(
    subject="024",
    task="med1breath",
    suffix="eeg",
    datatype="eeg",
    root=bids_root,
)
raw = read_raw_bids(bids_path=bids_path)
```

Tested with both stable and development versions of MNE and MNE-BIDS (you are running the stable versions, so it should work). Maybe the file is corrupted? The downloads from openneuro can be hectic.. Maybe try to download it with `openneuro-py` as I did?

jo780-full, 2022-11-15:
Can you please send what you did I would appreciate it .thank you

mscheltienne, 2022-11-15:
Install `openneuro-py` and use it to download the dataset (I first cloned in from the .git and the files were as often corrupted).

```
$ pip install openneuro-py
$ openneuro-py download --dataset=ds003969 --target_dir=/Users/scheltie/Documents/datasets/openneuro/ds003969/ --include=sub-024
```

Based on the documentation written in the README of `openneuro-py`: https://github.com/hoechenberger/openneuro-py
And then I loaded the data as you did with the python code snippet above.

mscheltienne, 2022-11-15:
I guess that the units have changed compared to older versions? I am not super familiar with this part, but I think it's linked to https://github.com/mne-tools/mne-bids/pull/1059
@hoechenberger I'll let you comment on this ;)

drammock, 2022-11-15:
when a channel type gets changed (in this case presumably from "EEG" to "Temp") then these warnings happen.  Same with GSR1 and GSR2, which are skin conductance (in Siemens).  All of them were apparently stored as "eeg" and MNE is changing their channel type based on the channel names, but it wants to make sure you know that in case it guessed wrong.

jo780-full, 2022-11-15:
Oh got it thank you

hoechenberger, 2022-11-15:
This issue should actually be transferred over to MNE-BIDS. I believe what's happening is in fact what @drammock said, except that the channel types aren't changed based on their name but based on the BIDS metadata. That is, I would think that we actually should **suppress** these warnings, as MNE-BIDS is doing exactly what it is supposed to do -- hence there is no need at all to emit a warning!