mne-tools/mne-python #11327: info['subject_info']['birthday'] should support a datetime object

hoechenberger, 2022-11-14:
### Describe the new feature or enhancement

Working on https://mne.discourse.group/t/minor-priority-issue-with-converting-mne-raw-into-bids-with-mne-bids-age-birthday-info-of-subject/5922, I realized that a subject's birthday must be specified as a tuple of ints. I believe this is inconsistent with the way we allow to specify the measurement date (datetime object).

### Describe your proposed implementation

It should be possible to specify `Info['subject_info']['birthday']` as a `datetime.datetime` object. 

### Describe possible alternatives

–

hoechenberger, 2022-11-21:
@larsoner Any opinion on that?

larsoner, 2022-11-21:
Agreed this makes sense. We can gradually transform these tuple-of-two-int into `datetime` as we find them.

PR #12720 (2024-07-17): ENH: Use date and datetime objects
1. Use `datetime.date` for `info["subject_info"]["birthday"]` rather than a year/month/day tuple
2. Use `datetime.datetime` for `info["helium_info"]["meas_date"]`
3. Fix in-memory anonymization for EDF by deleting `edf_info["subject_info"]` and `edf_info["meas_date"]`

Closes #11966
Closes #11327
closed 2024-07-17 by hoechenberger