mne-tools/mne-python #11342: spectral plot has no colors

ggrrll, 2022-11-29:
### Description of the problem

 I am plotting the spectra of few channels with `compute_psd().plot(spatial_colors=False)` 

I was expecting some colours, like in the 'midline' plot here
https://mne.tools/stable/auto_tutorials/raw/40_visualize_raw.html#plotting-spectral-density-of-continuous-data

does `mne` plot colours  only if there are spacial  information ? 
if yes,  this is not a bug, but rather a feature request

### Steps to reproduce

```Python
try to plot spectra of a `mne.io.RawArray` object
```

### Expected results

a colour for each channel

### Actual results

no colors

### Additional information

```
Platform:         macOS-13.0-x86_64-i386-64bit
Python:           3.9.13 (main, Aug  7 2022, 01:33:23)  [Clang 13.1.6 (clang-1316.0.21.2.5)]
Executable:       /usr/local/opt/python@3.9/bin/python3.9
CPU:              i386: 12 cores
Memory:           32.0 GB

mne:              1.2.2
numpy:            1.22.4 {unknown linalg bindings (threadpoolctl module not found: No module named 'threadpoolctl')}
scipy:            1.8.1
matplotlib:       3.5.2 {backend=module://matplotlib_inline.backend_inline}

```

agramfort, 2022-11-29:
yes to have colors you need channel locations. You can add these with the
set_montage method

> Message ID: ***@***.***>
>

ggrrll, 2022-11-29:
ok thanks -- however, it would be nice to relax this constraint (basically, it can assign  random positions, if they are not provided)

ps: the bug label can be removed of course

drammock, 2022-11-29:
In that case how will you be able to tell which color correspond to which channel? We won't *literally* assign random channel positions and then plot a head inset showing those fake positions --- that would be misleading.

larsoner, 2022-11-29:
In principle we could use a different color scheme/map (maybe just one uniform one, like viridis) that just colors by channel number/index. That would help differentiate lines while still making it at least somewhat clear that the colors are not spatial (since the colors gradients there indicate left/right anterior/posterior somehow IIRC). The name `spatial_colors` becomes a bit weird there but we could live with it. Something like `spatial_colors='order'` or something, where our default `spatial_colors=True` can mean "'positions' if available, 'order' otherwise" so that by default channels always have different colors.

drammock, 2022-11-29:
I guess I'd be OK with using viridis or plasma to assign colors based on channel sequence.  It's still a bit misleading because it implies that channels that are adjacent in sequence are more similar than channels that are not adjacent in sequence.  But I guess there's no way around that.

WDYT about a warning or info, like "spatial_colors requested but channel positions not found, falling back to sequential color map."  That way we don't have to support any new string-based param values.  I guess my assumption underlying that suggestion is that nobody would ever *choose* sequence-based coloring if they had location info available, so YAGNI the ability to specifically ask for it. Do you think that's right?

ggrrll, 2022-11-30:
glad to see that this sparked a discussion for FR :) 

@larsoner  got my point I guess

this is my current use case: we do not have positions, we are just testing different electrodes types, which are located very close to each other. So, to answer to @drammock we differentiate simply by the channel id (which is a meaningful name, related to the experimental condition)
Currently I am plotting the spectra 'by hand' (a bit of matplotilib code), where my legend is indeed made of channel names

agramfort, 2022-11-30:
rather than complicating these on mne side @ggrrll I would suggest you make up a default/random montage for your data. It is pretty trivial to do with 

https://mne.tools/stable/generated/mne.channels.make_dig_montage.html

my 2c

ggrrll, 2023-10-02:
hi, 

sorry for the late reply -- we stopped working  with that long time ago, an I am soon leaving the company

larsoner, 2023-10-02:
Okay I'll close then!

closed 2023-10-02 by larsoner