mne-tools/mne-python #11343: Default arguments of `raw.plot_psd()` changed?

cbrnr, 2022-11-29:
### Description of the problem

It seems like the default arguments of `raw.plot_psd()` changed in MNE 1.2 (see example below).

### Steps to reproduce

```Python
import mne
from mne.datasets import sample

fname = sample.data_path() / "MEG" / "sample" / "sample_audvis_raw.fif"
raw = mne.io.read_raw(fname).pick_types(eeg=True).load_data()
raw.plot_psd(fmax=60)
```

### Expected results

Result from MNE < 1.2:

[image]

### Actual results

Result from MNE >= 1.2 (including `main`):

[image]

### Additional information

Platform:         macOS-11.7.1-x86_64-i386-64bit
Python:           3.10.8 (main, Oct 13 2022, 10:18:28) [Clang 13.0.0 (clang-1300.0.29.30)]
Executable:       /Users/clemens/Projects/mne-python/.direnv/python-3.10.8/bin/python
CPU:              i386: 8 cores
Memory:           Unavailable (requires "psutil" package)
mne:              1.3.dev0
numpy:            1.23.5 {unknown linalg bindings (threadpoolctl module not found: No module named 'threadpoolctl')}
scipy:            1.9.3
matplotlib:       3.6.2 {backend=MacOSX}

drammock, 2022-11-30:
on `main` I see `Effective window size : 0.426 (s)` and on 1.1 I see `Effective window size : 3.410 (s)`.  Should be fairly easy to track down, I'll work on it.

cbrnr, 2023-01-11:
Unfortunately, this is still not fixed on `main` (the minimal example still reproduces the issue). Did something go wrong with merging #11345? @drammock do you have time to investigate?

drammock, 2023-01-12:
> Unfortunately, this is still not fixed on `main` (the minimal example still reproduces the issue). Did something go wrong with merging #11345? @drammock do you have time to investigate?

@cbrnr Are you sure?  I just ran 

```console
git switch main
git fetch upstream
git merge upstream/main
```
followed by
```python
import mne
from mne.datasets import sample

fname = sample.data_path() / "MEG" / "sample" / "sample_audvis_raw.fif"
raw = mne.io.read_raw(fname).pick_types(eeg=True).load_data()
raw.plot_psd(fmax=60)
```

and I get this figure:

[image]

cbrnr, 2023-01-13:
That works, but

```python
raw.compute_psd(fmax=60).plot()
```

doesn't. Since the note explicitly recommends to switch to this new variant, I'd expect them to yield identical results.

drammock, 2023-01-13:
OK, now I see the problem. Next time when you say 

> the minimal example still reproduces the issue

please check first. The code that is problematic for you is not in the minimal example at the top of this issue.

cbrnr, 2023-01-13:
Sorry about that, I missed that line.

PR #11345 (2022-12-01): fix default n_fft for spectrum plots
closes #11343 

there was already code in there to use the old default `n_fft` in hopes of avoiding this bug, but that code wasn't getting triggered in some cases because it looked for `method=='welch'` and was receiving `method=='auto'`.  Now we convert "auto" to a proper method name right away.
PR #11418 (2023-01-19): harmonize plot_psd() and compute_psd().plot()
closes #11343
closed 2023-01-19 by larsoner