mne-tools/mne-python #11360: [BUG] If T1 and T2 are used in recon-all, pial surface not found for `mne.viz.Brain`

alexrockhill, 2022-12-08:
### Description of the problem

The files are called `lh.pial.T1` and `lh.pial.T2` in the latest version of freesurfer.

### Steps to reproduce

```Python
Do a recon-all with a T2 and then do

import mne
brain = mne.viz.Brain(sub='1', subjects_dir='...')
```
```

### Expected results

Brain launches

### Actual results

```
FileNotFoundError                         Traceback (most recent call last)
Input In [26], in <cell line: 4>()
      6 info = mne.io.read_info(op.join(subjects_dir, f'sub-{sub}', 'ieeg',
      7                                 'ch_pos.fif'))
      8 trans = mne.coreg.estimate_head_mri_t(f'sub-{sub}', subjects_dir)
----> 9 brain = mne.viz.Brain(f'sub-{sub}', **brain_kwargs)
     10 groups = dict()
     11 for ch_name in info.ch_names:

File ~/projects/mne-python/mne/viz/_brain/_brain.py:356, in Brain.__init__(***failed resolving arguments***)
    353 geo = _Surface(self._subject_id, h, surf, self._subjects_dir,
    354                offset, units=self._units, x_dir=self._rigid[0, :3])
    355 # Load in the geometry and curvature
--> 356 geo.load_geometry()
    357 geo.load_curvature()
    358 self.geo[h] = geo

File ~/projects/mne-python/mne/viz/_brain/surface.py:124, in _Surface.load_geometry(self)
    122     coords[:, 1] *= -1
    123 else:
--> 124     coords, faces = read_surface(
    125         path.join(self.data_path, 'surf',
    126                   '%s.%s' % (self.hemi, self.surf)))
    127     orig_faces = faces
    128 if self.units == 'm':

File <decorator-gen-58>:12, in read_surface(fname, read_metadata, return_dict, file_format, verbose)

File ~/projects/mne-python/mne/surface.py:809, in read_surface(fname, read_metadata, return_dict, file_format, verbose)
    754 @verbose
    755 def read_surface(fname, read_metadata=False, return_dict=False,
    756                  file_format='auto', verbose=None):
    757     """Load a Freesurfer surface mesh in triangular format.
    758 
    759     Parameters
   (...)
    807     read_tri
    808     """
--> 809     fname = _check_fname(fname, 'read', True)
    810     _check_option('file_format', file_format, ['auto', 'freesurfer', 'obj'])
    812     if file_format == 'auto':

File <decorator-gen-0>:12, in _check_fname(fname, overwrite, must_exist, name, need_dir, verbose)

File ~/projects/mne-python/mne/utils/check.py:244, in _check_fname(fname, overwrite, must_exist, name, need_dir, verbose)
    241             raise PermissionError(
    242                 f'{name} does not have read permissions: {fname}')
    243 elif must_exist:
--> 244     raise FileNotFoundError(f'{name} does not exist: {fname}')
    246 return fname

FileNotFoundError: File does not exist: /home/alex/file/path/derivatives/sub-80/surf/lh.pial
```

### Additional information

Normal dev install.

larsoner, 2022-12-08:
It would be good to automatically look for `pial`, `pial.T1`, and `pial.T2` in whatever order is typically best (I have no idea what order that is) when someone requests `'pial'`

alexrockhill, 2022-12-08:
Probably the order you wrote above is reasonable. I don't have time to do this right this minute but I can soon.

PR #11361 (2022-12-21): [BUG][MRG] Fix surface finding for mne.viz.Brain
Fixes #11360
closed 2022-12-21 by drammock