mne-tools/mne-python #11370: DOC: Add resources for Python, terminal use etc.

britta-wstnr, 2022-12-15:
### Proposed documentation enhancement

We have tutorials on how to get started with MNE-Python that cover some more general aspects of Python (e.g. https://mne.tools/stable/auto_tutorials/intro/15_inplace.html). 
I often get asked where to start if you do not know any Python at all yet - and as far as I am aware there is no material on the MNE homepage for that.  
Maybe we could add a tutorial page with resources we can recommend for getting started with Python, numpy, or also the use of a terminal.

drammock, 2022-12-15:
We have this page https://mne.tools/stable/overview/learn_python.html

What I'm hearing is that it's not discoverable.

britta-wstnr, 2022-12-15:
Oh! 
Well, I searched for it and did not find it. Based on the questions I get asked, it seems like it is the same for users - and it seems @larsoner was not aware either ... 

Maybe we could move it to tutorials? This is where most people seem to land if they feel like they need an in-depth start.

larsoner, 2022-12-15:
A few potential ideas

1. Move from `overview` to tutorials
2. Add a new tutorial `intro/05_new_to_python.py` that just links to the overview doc
3. Add to existing `intro/10_overview.py` a `.. sidebar::` or so that links to the overview doc

drammock, 2022-12-15:
IMO the docs need a medium-sized overhaul, cf #10707.  I think this issue and that one should be addressed together, as I don't think we can find the "right place" for this info in the current (IMO wrong) organization of the (non-tutorial) docs content.